aenniw/ARDUINO #3: Looks great.

iLLiac4, 2018-09-07:
Hi. Just yesterday I have bought two of those tables.
Can you upload some videos of the speed and noise and add a little more pictures of electronics inside the case to see the soldering connections.

TNX

aenniw, 2018-09-17:
Hi sorry for late response... I will update pictures, schemas and stl models during this week. Later on also publish prebuilded binary...

iLLiac4, 2018-09-18:
Oh. This is great. So we should wait for newer updated stl files before starting to print parts?

TNX

aenniw, 2018-09-19:
Yes I would wait but if you already printed them, then you can use the old one only issue that you can experience may be loose screw holes and arduino-nano mount.... Currently I'm reprinting those parts on mk3 to make sure dimensions for those holes are ok as my old printer was quite  sloppy in printing holes and small parts... 

Currently the changes are:
- downsize screw-holes from 0.16 to 0.14 radius (pre-drilling may be needed for some printers)
- widen arduino-nano, power adapter holes so that more robust connectors can be connected 
- increased pin height for buck-converter for ones with thicker PCB

But I will add change log for STLs and code when I'm done with reprinting ....

iLLiac4, 2018-09-23:
I have already got some parts from ebay. I think I have figure out the positions for the components. For the two relay board the legs in stl are a little bit too wide for the relay I have got.
I have printed gt2 wheels with ABS and the look good. Maybe 2-3% scale down will do fit tighter on rotary encoder.
For some parts I can not figure where they should fit.
Hope you will soon upload the rest of the things since I hope this week I will gather all the parts.

The case was printed in PLA and with 0,2mm resolution. It took 24 hours to print.

aenniw, 2018-09-25:
24h is quite long I've done mine 0,2mm in 10h on mk3 and even on Anet A8 it took around 12h

I have updated STLs with changes mentioned above and things that you mentioned, also added more detailed how-to assemble each parts with pictures and added videos showcasing its functionality.
However binary will be releasees after Issue #5  is resolved as it unexpectedly restart Arduino.

closed 2018-10-01 by aenniw