aenniw/ARDUINO #10: Coupler not strong enought

iLLiac4, 2018-10-15:
I have used coupler printed in ABS and it was destroyed immedietly. The 6mm hex key just spinned inside.
I think a metal part one shoud be used something like https://www.ebay.com/itm/2-3-4-5-6-7-8mm-Aluminum-Flexible-Shaft-Coupling-Rigid-Coupler-Motor-Connector/291882575832?hash=item43f58ea7d8:m:mJ7ZSkXCSVnDjsGM3CKchsQ:rk:5:pf:0
and only printed part would be the GT2 ring.

dabde, 2018-10-19:
Hi,

I was also thinking about another solution, but ABS is not the best for this case. PLA/PETG are more ridgid/stiff, to not get deformed. It's take more momentum but will crack if the stress is exceeding. ABS is more, "I don't crack, but I soft deform me on a medium load".

iLLiac4, 2018-10-19:
I have bought the coupler above 6-7mm and have modded the gt2 wheel with 14mm hole and printed it.
I am still waiting for a coupler but it will be build like tank and will not deform.
I can upload moded GT2 wheel but it is easy fix with fusion360. I have used pulley gt2 wheel to mode it.

aenniw, 2018-10-21:
Hi
I have printed coupler in PLA with 100% infill and it works, only time when it broke was when I hit max/min table height but other than that it holds quite good.

@iLLiac4
If you get it working with that metal coupler feel free to open PR or let me know to update STLs and instruction as that would be much better as the one currently used...

iLLiac4, 2018-10-21:
Sure will do that.

closed 2018-11-12 by aenniw