aenniw/ARDUINO #7: Bug report

iLLiac4, 2018-10-15:
I have finally made it and it works at least on the bench for now :)
There is a problem if you have for example pos 1, 2 and 3 set.
If you press position 3 and the motor moves and then during the moving of the motor you press button UP the the display resets and it shows the reset position and not the real one. This way you loose the calibration.

The same happens if you go to position 1 and then during the moving presses the low button. It resets the display and also relays and shows the position of the set 1 and not the actual one.

Also it  would be useful to cancel the movement during the call preset.
For example if i select number 3 and the motor starts the movement could be stopped with pressing any button. No matter if i press up, dow, 1, 2 or 3 the first press during the movement of the button should stop the motor.

Other-way it is very well documented.

I think that on frittzing schema there is one error. The clk and dio of the rotary encoder should be reversed. I am not 100% sure but will check for sure. So D3 and D2 on arduino should be connected in reverse order.

aenniw, 2018-10-21:
Hi 
I have updated the code-base quite a bit #12 could you recheck those issues once more agains current master if possible? I will try to look onto them during next week or so...
Thanks

aenniw, 2018-10-28:
Movement cancelation feature adressed in #16
Encoder wiring seems ok based on pin defininitions in code...
Board reset has its own issue #14

closed 2018-10-28 by aenniw