aenniw/ARDUINO #14: New bug

iLLiac4, 2018-10-21:
Hi I have tested your new code and there is a bug. You can not set memory on keypad 1. It goes always to 0000.

aenniw, 2018-10-25:
Hi I have added EEPROM clear on first start and updated few things that could case this issue could you recheck current `master` if the issue still persists? Thanks

iLLiac4, 2018-10-25:
Hi. I have tested the new version and the presets seems to work.
But there is another problem.
If you store the position to one of the presets and you call it the motor starts to spin. And if you press during the movement the button up or down the arduino resets and the last position is lost and with them the calibration. I hope you know what i mean.

Yust call a preset and during movement press button up or down. On of those presses will reset the arduino.

It would be nice if you do press any button during the movement of the motor, the motor should just stop.

aenniw, 2018-10-28:
Does the reset occurs if you power arduino via mini-usb? Note that while testing it you need to disconect arduino from internal buck-converter...

iLLiac4, 2018-11-07:
Fixed in latest release so I am closing it.

closed 2018-11-07 by iLLiac4