aenniw/ARDUINO #20: Slow motor start

iLLiac4, 2018-11-09:
According to the converstaion:
---------
Q:
Commercial version of Ikea motor desk starts motors slowly. When you press the button up or. down it does not give full power to the motros but if stars them slow to avoid motor shocks and then accelerates to its full potentional.

A:
Regarding slow start of the motor with current setup with relays its not doable, but with something like BTS7960B H-Bridge it could be done... Could you open an separate issue regarding PWM motor control and I will look into that when get my hands on that H-Bridge?
--------

iLLiac4, 2018-11-09:
Ebay link for possible working part:
https://www.ebay.com/p/Double-Bts7960b-DC-43a-Stepper-Motor-Driver-H-bridge-PWM-for-Arduino-Smart-Car/1062875951

iLLiac4, 2019-01-05:
Hi.
I have not got the H-bridge yet to start testing. But I have still few questions about h-bridge.
1. Will you update the frizing schema to adopt h-bride?
2. Will you update stl models so that they will be compatible with h-bridge?
3. Is it possible to make already printed stl models compatible with h-bridge? (Drill and screw some box on already printed object)?
4. Is a board with two relays still needed after h-bridge implementation?
5. Please keep both options open also in the future; one as it is now and one with H-bridge option.

TNX

aenniw, 2019-01-06:
I've recieved this week one too, but as I haven't access to **3d printed** I could not start working on it... Currently I've just pushed Initial impl that needs to be tested #27 and today I should finish case design on it ...

Regarding your questions:
1. Yes
2. Yes, I will add new case for h-bidge setup
3. That will be posible but with quite a lot of effort/trial-error I would say as the PCB and the heatsink is quite bit, I'll add an PR with new case model...
4. No it's not I guess but I'll would like to keep both of them as version with **Relays** is cheaper
5. Yep that's the plan for both the firwmeare and STLs

closed 2019-01-15 by aenniw