aenniw/ARDUINO #26: Few notes after a mont of use.

iLLiac4, 2018-12-17:
I have been using this system now for a month and I have few suggestions:
1. Finding the right power supply is very hard. I have tried many and not all works the same. You can use also 19v one but some laptop ones has protection integrated.
One good option is to use old computer power supply. I would recomend to use something like this board https://www.ebay.com/itm/ATX-power-supply-adapter-changer-module-XH-M229-desktop-pc-board-24pin-OJBILU/183574137281 to easy hack it.
To stop power supply fan from spining all the time you could use https://www.ebay.com/itm/KSD9700-Degrees-Celsius-Thermostat-Control-Temperature-Switch-5A-250V-40-50/302834915986 to start the fan only when tepereture rises above 45 or something.
As a side use you can use the same power supply as a bench power supply. And with a decent voltage regulator you can make it even more versatile: https://www.banggood.com/custlink/333GKfKhHD

2. The screws I have used are
- for control pannel: 4,0x16
- for motor box: 3,0x20

aenniw, 2019-01-07:
Currently I'm using 12V - 3A PSU from meanwell and as I dont have many things on table it is enought.

**Model NO:** [_GS36E12-P1J_](https://www.meanwell-web.com/en-gb/ac-dc-wall-mount-adaptor-output-12vdc-at-3a-input-gs36e12--p1j)
**INPUT:** _100-240V~, 50/60Hz, 0.7A_
**OUTPUT:** _12V - 3.0A, 36W Max_

WIll add more links regarding PSU to docs and screw dimensions...

aenniw, 2019-01-07:
Updated in #30

closed 2019-01-07 by aenniw