aenniw/ARDUINO #31: H-Bridge stl small modification.

iLLiac4, 2019-01-15:
I saw new models for h-bridge.
Can you add a little more space between keypad connector and dc jack.
This way the dc jack hole can be moded if there is more space.
For example I am using thise jacks https://www.ebay.com/itm/10pair-12V-3A-2-1x5-5mm-Male-Plug-Female-Socket-Panel-Mount-Jack-DC-Connector/273345758958?epid=17021273616&hash=item3fa4ad46ee:g:iYIAAOSw2FVbPxry:rk:46:pf:0
and I have to mod a hole a little.
ATM there is not enought space between dc connector and keypad connector to mod it.

P.S.
What is the purpose of the sqare on h-bridge model beside arduino mount?

TNX

iLLiac4, 2019-01-16:
Much better. It would be easier to mount if the hole would not be so near the wall.
Something like:
[
[image]
](url)

iLLiac4, 2019-01-17:
I havw checked your manual and saw the picture of the botton H-Relay case. I do not know if the H-Bridge get hot but, if it does then I think the space between case and heatshing of H-Bridge should be increase to avoid melting the plastic especialy PLA which is not heat resistant.

aenniw, 2019-01-17:
Updated connectors spacing, on daily basis the heatsink isn't even warm so frame melting won't be an issue I think, but later this week I will try some load tests to verify it...

closed 2019-01-17 by aenniw