aenniw/ARDUINO #32: Urgent - Design change needed

iLLiac4, 2019-01-15:
Today my motor has burned. Funny is that it has burned when the table was lowering, so no max current was used.
I have smelled something and then saw smoke from the motor box. I have quickly unscrew it and the motor was extremely hot and plastic was melting and smoking. I really do not know the cause why this has happened. Maybe the wires did  not stand the high current and over time melted and cause the short of the motor.
I will definitely use silicone wires when I fix this.
Here are some pictures with motor already disassemble.
https://www.dropbox.com/s/4k363yrko571gem/20190115_073516.jpg?dl=0
https://www.dropbox.com/s/zhrmst5eu32usds/20190115_073553.jpg?dl=0
https://www.dropbox.com/s/2wht3zi3zr6kd62/20190115_080947.jpg?dl=0
https://www.dropbox.com/s/orfj8cq49ppir36/20190115_073524.jpg?dl=0

Any way I think the design has to be changed. Some kind of external accessible fuse should be implemented or. if you have in mind something better to protect the motor and wires against shorting or over current protection and to avoid burning.

I know some power supply has short circuit protection integrated, but this one obviously did not have one. I thing it would be wise to integrate one into the box.

iLLiac4, 2019-01-17:
I see that you have already started to implement protection under watchdog. Will you merge it or. it is not yet production ready?

TNX

aenniw, 2019-01-22:
Hopefully by the end of week it'll be merged with new models including holes for fuse holders...

iLLiac4, 2019-01-28:
I have tested the new feature of watchdog and it works but not enought to prevent motor burning i think.

To test it i have removet the gt2 belt so there were no rotay encoder connected. After up, down or preset it started to spin but then after a few seconds it stopped with err, which seems ok.

Then i have put the motor into the table, so  with load and still no gt2 connected and it actually protection did not kick in at all. The table was lifting and lowering without gt2 belt connected and did not reporty any error.
I think the error should kick in in max 0,2 - 0,5 second if there is no signal from rotary encoder and shuts down the motor.

aenniw, 2019-01-29:
Updated the thresholds for watchdog could you recheck if the watchdog triggers soon enough?

iLLiac4, 2019-01-29:
I have tested the settings in watchdog and have found that those works best for me under load and triggers if something is wrong.
#define WATCHDOG_TIMEOUT 10
#define WATCHDOG_DEADLOCK_CHANGE 1
#define WATCHDOG_OTHER_CHANGE 5
#define WATCHDOG_TOLERANCE 5

This code should be changed when testing without load because it will trigger err-2.
So comment like:

#define WATCHDOG_TIMEOUT 10 ; without load change to 2000
#define WATCHDOG_DEADLOCK_CHANGE 1 ; without load change to 10
#define WATCHDOG_OTHER_CHANGE 5 ; without load change to 50
#define WATCHDOG_TOLERANCE 5 ; without load change to 10

P.S.:
- A useful feature would be a restart option (if err.1 or. err.2 occurs) if the up and down button would be pressed at the same time for 3 seconds. 
- Also was noted that buttons up and down now works in reverse order. Up lowers the table and down rises it. Or. is this just by me and i have to change pins on arduino. But it is strange since i did not change anything. And also when moving down the leds number rises and vice verse. Something is definitely wrong with positions.

iLLiac4, 2019-01-30:
Hi. Yes it seems that something is wrong on the motor that i have rebuild.
I have tested it with the working one and everything seems to work. So just ignore the up, down eroor in previous build.

Can you comment what those changeses mean? I have just played with them to come to those settings that works ok for me.
Also settings seems ok that i have mentioned:
#define WATCHDOG_TIMEOUT 10
#define WATCHDOG_DEADLOCK_CHANGE 1
#define WATCHDOG_OTHER_CHANGE 5
#define WATCHDOG_TOLERANCE 5

I have tested this on relay version. I have still not received h-bridge. Do you think that h-bridge settings have to be different?

Also a reset of arduino when err.1 or err.2 appears with three seconds up and down simultaniously press would be nice.

aenniw, 2019-01-30:
Will add more info regarding those watchdog settings... And the H-Bridge version should have different values as it scale it's speed at the beginning... So I will provide different settings per build...

aenniw, 2019-02-02:
Updated docs with fuse info, new models with fuse holder and separate watchdog settings for RELAY build.

 I would like to keep the manual power cycle of the board, so that you need to check the table if things are OK, not only unblock it via button press and hit the issue once more...

closed 2019-02-10 by aenniw