aenniw/ARDUINO #43: H-Bridge startup parameters

iLLiac4, 2019-02-26:
Which part of the code for H-Bridge is responsible for the startup control of the motor?

I would like to tweak it so the motor will start to work at full speed quicker than at the moment.
ATM it takes almost 3 seconds to achieve the full throttle. And because it is so slow it falls into err.1.
Totally enought would be already 0,5 sec.

aenniw, 2019-02-27:
Decreasing the **SPEED_STEP_DURATION** in [MotorBridge.h](https://github.com/aenniw/ARDUINO/blob/6004b4ebaa20f23a01ae9f06a4883a93a4a82920/skarsta/lib/motor-bridge/MotorBridge.h#L9) should do it.

iLLiac4, 2019-02-28:
Great tnx. i have found SPEED_STEP_DURATION 30 best value ATM. Will do also some further tests on the bigger desk during the weekend.

closed 2019-03-04 by aenniw