aenniw/ARDUINO #48: Can you please share the SRC for the models?

kquinsland, 2019-05-06:
Hi there. I am grateful that you have put this together and made it a complete DIY project.

I was wondering if you'd consider sharing the raw source / CAD files that you used to render the `.stl` files. I would like to "remix" this project and several of the modifications that i have in mind will be rather difficult if i have to edit the stl files.

Please release the CAD files behind the stl files :).

aenniw, 2019-06-15:
Hi I do not plan on releasing the files for now, however reach me via email I will send you what I have hopefully that will help you... (Note that source files are `c4d` as I mainly use Cinema4D)

closed 2019-06-15 by aenniw

sandman0, 2020-02-08:
Sorry to dig out an old (closed) thread. But I was about to create a new issue when GitHub suggested there is a similar existing issue, so here it is.
Firstly Martin, this is fantastic work, thank you so much. I have built it and am using it for the last couple of months. Now, I want to make two changes:
1. Use something else than the rotary encoder for positioning. You have already done some work for this but I haven't tried it yet. I may go the ultrasonic sensor way.
2. It happened a couple of times that the hex rod slipped off from the motor shaft and it was hard to fix with the current way everything is mounted in a single box. I want to move the motor out of the main electronics box.

For 2, it will be great to get my hands on the cad sources. Would it be possible for you to export your c4d as OBJ and send them to me? It will be greatly appreciated!
Thank you.

aenniw, 2020-02-24:
Sure, can you ping me on email?

sandman0, 2020-03-28:
Sure, what is your email id?

aenniw, 2020-03-31:
Sry didn't noticed that I had private email in GH acc :laughing:, now it should be vissible...