aenniw/ARDUINO #49: Loosing steps

iLLiac4, 2019-05-12:
It looks like that somehow it losses steps and this causes to get decalibrated over time.
I saw this when I have lowered the table and it looked that it was getting always more lowered than it was after the initial calibration.
Eventely over time it would hit hard limit and burned the fuse.
Maybe the rotary encoder losses counts or arduino is not able to count it right.
And it is not losse GT belt.

aenniw, 2019-06-15:
I guest that 2 things could do it:
 - Button handling is done via interupts
 - Rotary encoder precission is to low

I will try to update `FIRMWARE` to use Non Interupt handling of buttons addressed in `feature/NI-skartsta` and try to migrate rotary encoder to one of these optical count sensors [LM393 H2010](https://www.ebay.com/itm/LM393-H2010-Photoelectric-Infrared-Radiation-Count-Sensor-Module-For-Arduino/264247290987?_trkparms=ispr%3D1&hash=item3d865dac6b:g:SZYAAOSwizpckeex&enc=AQADAAAB0KX%2FKt4E1xf3SDqEdBclaYZ9%2B4o9IVf2ShUi78Fh6J10NVAcktunyxCfUyFsuHD%2FIRQBatap62icHh%2Fzzrl8Gapw%2FLjAZqfoknF20wY39J8Drsxm6wOkPcO46pYT%2F2AR3NpqDJdMZ624ms7SGD9oLskFGs7a976NlHNTDKk3606jUwTRI2JhoRSBBvDqDCyMAVhGyLEQM05mVWN0UOvjCxCcZvnlgI71crTknlBfvBRV8DtA7sdvQ5vfkVPbCX3HwnVXKp68WoQxo8QPQLVqvpZH8bKj%2FTyXe%2B7xOq%2F1woWKQfD8RItnT6ZdUWXzpmB6MUdGriaDgB%2BPvmrFPgKftJu9TqKSFcF9KrPW1zowQuke7fgfvBMYaV8UskOZMVaYl7%2FVzkuCtKxxi%2FrH9wr7KQlwtRrKahcDsHDt%2Blyb4dL2WLPPmT%2B2IhEx7nOYLCjX7fjYrGUjR2AZEOw8psRFxlGTV4L%2FDHbgPeb2JwhcQIEloo%2B94Py7lpZxMnrcTrtWkLae0uncNXtLxFjc1KVwp%2FH4si8PfN8sicjXRmASEugu%2Fmj3yussviYuOAam2yy43%2FHcvpDs%2FUjIBmvB5kynSIDr7V%2FADDTDx1Imoc2QLwQI&checksum=2642472909879e6c8905545d4a4bb7790a55365f4f9a) [LM393](https://www.ebay.com/itm/5PCS-Slot-Type-IR-Optocoupler-Speed-Sensor-Module-LM393-for-Arduino/132478779559?hash=item1ed859c0a7:g:disAAOSw0lVbCTU0)

iLLiac4, 2019-06-16:
Have tried the new version with h-bridge.
I had to change a few other parameters to get rid with err.1 with new version.

-----------
#include <Arduino.h>
#include <Motor.h>

#define MIN_SPEED 100
#define MAX_SPEED 255
#define SPEED_STEP_DURATION 15
----------

and

-----------
#ifndef H_BRIDGE_MOTOR
#define WATCHDOG_TIMEOUT 100
#define WATCHDOG_DEADLOCK_CHANGE 2
#define WATCHDOG_OTHER_CHANGE 5
#define WATCHDOG_TOLERANCE 5
#else
------------

Now will test it a few days to see if it will loose steps.

iLLiac4, 2019-06-24:
Hi. I have tested it and it works better then it was. It is still not perfec but much improved. Have tested only with H-bridge.
I think you can merge it.

iLLiac4, 2019-07-03:
Hi. I can see that you have made some commits. The one thing I have noted is that one version of the gate ring will have to be little modified so that it will be fit on optional aluminium 6-7mm shaft adapter.

Aslo can je please tell me which opto sensor have you order so i can already buy it because it takes time to arrive. TNX

aenniw, 2019-07-06:
Regarding opto sensor currently I'm testing [Optocoupler ](https://www.ebay.com/itm/IR-Infrared-Slotted-Optical-Speed-Measuring-Sensor-Optocoupler-Detection-Module/173695686874?epid=859550472&hash=item287111d8da:g:zNAAAOSwAx9cF53p) as the previous one were quite nisy and needed a lot of debouncing for some reason... However I would wait until #54 is merged

aenniw, 2019-10-21:
So the **opto coupler** was even more unreliable so now I'm testing [hall encoder](https://www.ebay.com/itm/25mm-Magnetic-Hall-Encoder-11PPR-Double-Phase-Output-Square-Wave-For-370-Motor/153185900651?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D40719%26meid%3Dd958f0c6018f42fda7a911a5e348da71%26pid%3D100675%26rk%3D2%26rkt%3D15%26sd%3D283429506020%26itm%3D153185900651%26pmt%3D0%26noa%3D1%26pg%3D2380057&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3Ab582482f-f43c-11e9-ad2d-74dbd180db7e%7Cparentrq%3Aefe360dd16d0aa18f1558095ffeb2fe9%7Ciid%3A1), it seems better, however it would be best to find motor with embeded **hall encoder** as this one that I've found needs to be customized a lot ...

Also I've made som changes to firmware that could affect precision so check them out if you have time.... branch `feature/hall-sensor`

iLLiac4, 2019-10-29:
Is this hall-sensor branch compatible also with regular rotary encoder?

aenniw, 2019-10-29:
Yes It should be

iLLiac4, 2019-10-30:
I have tested it and it is even worse :( it looses steps faster.

aenniw, 2020-01-07:
Try alternative distance measuring #67

aenniw, 2020-01-20:
Testing hall-encoder [feature/hall-encoder](https://github.com/aenniw/ARDUINO/tree/feature/hall-encoder/skarsta)

aenniw, 2020-05-12:
This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days

closed 2020-06-19 by aenniw