aenniw/ARDUINO #59: opto skarsta

iLLiac4, 2019-08-03:
Hi.

Have you already tested optical sensor?

aenniw, 2019-09-24:
Hi sorry for delayed response, I have ended with [IR module](https://www.ebay.com/itm/IR-Infrared-Slotted-Optical-Speed-Measuring-Sensor-Optocoupler-Detection-Module/173695686874?epid=859550472&hash=item287111d8da:g:zNAAAOSwAx9cF53p), other game quite a lot of false reading despite nothing was moving... however I'm still strugling with "loosing" steps as movement down generates less tick...

During next few weeks I'll have more time  to look at it ...

aenniw, 2019-10-21:
Closing this as even with multiple IR modules with offset detection seems to bee even more unreliable... will look into hall sensor

closed 2019-10-21 by aenniw