aenniw/ARDUINO #61: H-BRIDGE

p0thi, 2019-09-14:
Hi everyone,
I have a question about the power output of the H-BRIDGE: When the speed is set to 255, the output voltage is just 3v and not 12v like the input. And because of that the motor spins with only about ~60RPM at max speed.
Am I doing something wrong, or is that expected behaviour?

Thanks for the help! :)

p0thi, 2019-09-16:
I solved my problem:
I am using a NodeMCU board with an ESP8266 chip for my project. Unlike the arduino boards this chip uses 10 bit as the PMW resolution.
I changed the code to be able to handle a 10 bit speed value and now it runs at the expected speed at 1023. [Question on StackExchange](https://electronics.stackexchange.com/questions/458722/how-to-get-right-voltage-output-with-bts7960b-h-bridge/458736#458736)

closed 2019-09-16 by p0thi

RikMuilwijk, 2021-04-18:
@p0thi great idea to use a NodeMCU, could you share your code? Would love to adapt this project using ESPHome.

p0thi, 2021-05-04:
I eventually stopped trying to use my NodeMCU for this project, because it has not enough pins.
I even tried to increase the number of pins with multiplexers, but that did not work for me.

Instead I wrote a new Service to enable serial communication with the Aruino Nano. You can find my code, my config, and how I use it with ESPHome for my Home Assistant in this PR #137 
So I actually use a second micro controller with WiFi (Wemod D1 mini) to communicate with the Arduino Nano by using serial communication.
I think it would not be hard to use a NodeMCU instead of the Wemos board.

RikMuilwijk, 2021-05-04:
Awesome! Thanks, you just saved me a ton of work, can’t believe I missed that PR.

p0thi, 2021-05-05:
No worries, you didn't miss it. I created it just yesterday. I had this code laying arround for over half a year. And you inspired me to finally share it. :)