aenniw/ARDUINO #65: ArduinoIDE

TigerClaw33, 2019-12-11:
Hello, 

I installed all libs and also ArduinoSTL for the vector because "#include<vector>"was not working. 

But I get this error:

```

Arduino: 1.8.10 (Windows 10), Board: "Arduino Nano, ATmega328P"

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::Display(unsigned char, unsigned char)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::Display(unsigned char, unsigned char)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::set_blink(bool)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::print(unsigned int)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::print(char const*)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::set_brightness(unsigned char)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::cycle()'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::light_up()'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\display\Display.cpp.o (symbol from plugin): In function `Display::Display(unsigned char, unsigned char)':

(.text+0x0): multiple definition of `Display::disable(unsigned char)'

sketch\Display.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\service\Service.cpp.o (symbol from plugin): In function `get_period(unsigned long, unsigned long)':

(.text+0x0): multiple definition of `get_period(unsigned long, unsigned long)'

sketch\Service.cpp.o (symbol from plugin):(.text+0x0): first defined here

[... 497 lines cut ...]
Mehrere Bibliotheken wurden für "PinChangeInt.h" gefunden
 Benutzt: D:\Arduino\libraries\PinChangeInt-master
Mehrere Bibliotheken wurden für "Rotary.h" gefunden
 Benutzt: D:\Arduino\libraries\Rotary
Mehrere Bibliotheken wurden für "TM1637.h" gefunden
 Benutzt: D:\Arduino\libraries\TM1637
 Nicht benutzt: D:\Arduino\libraries\Grove_4Digital_Display-master
Mehrere Bibliotheken wurden für "Keypad.h" gefunden
 Benutzt: D:\Arduino\libraries\keypad
Mehrere Bibliotheken wurden für "Motor.h" gefunden
 Benutzt: D:\Arduino\libraries\motor
Mehrere Bibliotheken wurden für "NIButtons.h" gefunden
 Benutzt: D:\Arduino\libraries\ni_buttons
Mehrere Bibliotheken wurden für "PMButton.h" gefunden
 Benutzt: D:\Arduino\libraries\PolymorphicButtons-master
Mehrere Bibliotheken wurden für "Watchdog.h" gefunden
 Benutzt: D:\Arduino\libraries\watchdog
exit status 1
Fehler beim Kompilieren für das Board Arduino Nano.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

```

iLLiac4, 2019-12-11:
To compile it you have to use plaformio.

TigerClaw33, 2019-12-11:
Normally it also has to work with the Arduino IDE :(

TigerClaw33, 2019-12-11:
`ArduinoSTL @ 1.1.0 has been successfully installed!
LibraryManager: Installing Rotary
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\admin\.platformio\platforms\atmelavr\builder\main.py", line 155:
    target_elf = env.BuildProgram()
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\admin\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "c:\users\admin\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\admin\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "c:\users\admin\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 46:
    src_dir, remote_url, tag, silent
  File "c:\users\admin\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 61:
    self.check_client()
  File "c:\users\admin\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 127:
    "Please install Git client from https://git-scm.com/downloads"
================================================================================================== [FAILED] Took 1.56 seconds ==================================================================================================

Environment           Status    Duration
--------------------  --------  ------------
nanoatmega328-relays  FAILED    00:00:00.857
nanoatmega328-bridge  FAILED    00:00:01.563
============================================================================================= 2 failed, 0 succeeded in 00:00:02.420 =============================================================================================`

TigerClaw33, 2019-12-11:
`lib\display/Display.h:13:26: error: 'BRIGHT_HIGH' was not declared in this scope`

Why nothing is working????

iLLiac4, 2019-12-13:
Hi. I have tried to compile it now and it has also failed first. But first you have to download the whole zip file arduino and extract it https://github.com/aenniw/ARDUINO

The you have to  upadate also this one as stated in your log file "Please install Git client from https://git-scm.com/downloads"

Also run the Visual studio as administrator. And when compiling, compile only one version. Comment out the other one not used.

For example in platformio.ini

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html
[common]
build_flags = -O2 -std=c++11 -Wall -D__EEPROM__
lib_deps =
    ArduinoSTL
    https://github.com/brianlow/Rotary.git#master
    https://github.com/JCWentzel/PolymorphicButtons.git#master

;[env:nanoatmega328-relays]
;platform = atmelavr
;board = nanoatmega328
;build_flags = ${common.build_flags}
;framework = arduino
;lib_extra_dirs = ../lib
;lib_deps = ${common.lib_deps}

[env:nanoatmega328-bridge]
platform = atmelavr
board = nanoatmega328
build_flags = ${common.build_flags} -D__H_BRIDGE_MOTOR__
framework = arduino
lib_extra_dirs = ../lib
lib_deps = ${common.lib_deps}

TigerClaw33, 2019-12-13:
Hi, no it seems to be working . thanks. 

I tried it wihtout OO-programming with the Arduino Editor and it's also easy to implement all functions without so many libs. But your code is done and tested so easier to use. Thanks :)

closed 2020-01-07 by aenniw