aenniw/ARDUINO #66: Rotatory encoder and motor not working

TigerClaw33, 2019-12-23:
Hi, 

I am using a h bridge and conntected everything like you. 

For testing I conntected first only one button. But Display shows me "----". And if I push and hold the button for motordriving it drives one or two seconds and stopps. Than it shows me "err 1". If i turn rotatory by hand nothing happens. Display shows me "----". Rotatory is working. I tested it with other application.

I have a 5 Pin encoder and conntected only CLK and DATA two arduino. The SW pin is not conntected. I think it's for pushing encoder button.

iLLiac4, 2019-12-23:
---- it means that it is not calibrated yet. To activate calibration you need to use tilt button.
err 1 is shown when there is too much load.
err 2 is shown usually when there is some free spin. For example if you rotate the motor without load.
All those parameters can be set in the code. Check closed issues.
https://github.com/aenniw/ARDUINO/issues/46

Sw pin is not used.

TigerClaw33, 2019-12-24:
ok, rotatory is working now but I always get err 1. Motor is not loaded, it's on my desk for testing. If i press the button motor stopps after 1 oe 2 seconds. NOt loaded :(

iLLiac4, 2019-12-24:
I am sure it is ok. Insert them into desk so there is load. Those checks were implemented to avoid burning the motors, free spin,...

iLLiac4, 2019-12-24:
And gt2 belt has to be conected to the motor and rotary encoder. If not then err1 kick in of course.

aenniw, 2020-01-20:
Try to adjust one of 
```c
#define WATCHDOG_TIMEOUT 100
#define WATCHDOG_DEADLOCK_CHANGE 2
#define WATCHDOG_OTHER_CHANGE 5
#define WATCHDOG_TOLERANCE 5
```
in [Watchdog.h](https://github.com/aenniw/ARDUINO/blob/master/skarsta/lib/watchdog/Watchdog.h)

as your table may need different settings...

aenniw, 2020-05-12:
This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days

closed 2020-06-19 by aenniw