aenniw/ARDUINO #67: Loosing steps possible solution.

iLLiac4, 2019-12-24:
Hi.

I was searching for some hacks on ikea desks and found an interesting article claiming that it is 1cm accurate.
This is the article:
https://www.henrirantanen.fi/2016/04/18/smartdesk-sensor-upgrade/?utm_source=youtube.com&utm_medium=referral&utm_campaign=video&utm_content=IKEA+Bekant+SmartDesk+with+Arduino

Using echoing sensor it could replace rotary encoder.
For example this one https://www.ebay.com/itm/1PCS-Ultrasonic-Sensor-Module-HC-SR04-Distance-Measuring-Sensor-for-arduino-SR04/193011377769?hash=item2cf05fe669:g:~d8AAOSwW4Bd6f5J

What do you think would this be an option. I think it is possible but you are the master coder :)
And it this works it will always work and will not loose steps anymore.

iLLiac4, 2019-12-24:
There is a library for this sensor with even possibilities to improve the measurment further with dht22 or. dht11. But I doubt that this will be needed :)

https://dronebotworkshop.com/hc-sr04-ultrasonic-distance-sensor-arduino/

iLLiac4, 2019-12-25:
Just yesterday I have bought for a good price one stand desk and while searching for a hack to add controls to the desk I have found another use of ultrasonic sensor:
https://nortal.com/blog/cloud-controlled-work-desks-using-arduinos-node-js-polymer/

I am almost sure that this can solve the loosed steps solution with replacing the rotary encoder. What do you think?

aenniw, 2020-01-07:
Thanks!

I will look into that after I obtains needed parts a test it a bit...

aenniw, 2020-01-20:
Tried to  adapt to ultrasonic distance measuing, however sensor precission was around +-1cm so the improvement istn;t that big at the moment. Also I suspect that precission will be different based on floor type (glossy wooden floor or capet)...

Will look into it next weekend more...

iLLiac4, 2020-01-20:
Yes I know that the precission is at around form 1 to 2 cm. But i think this is ok. The point is that it can not decalibrate since it always measures. With rotary it was decalibrating every time more and more until it was so much decalibratet that it would hit hard limit and burn motors or. fuse.
With echoing it would always be at around 1-2cm distance and will not decalibrate.

aenniw, 2020-01-20:
Ok so you mean to use it only for hard limits of table?

iLLiac4, 2020-01-20:
No since 1 cm is in my opinion enought I was planing to use it instead of rotary encoder.

iLLiac4, 2020-01-20:
The calibration process could be the same as with rotary encoder. Maybee only it can display mm or. inches if choosen in code before compiling. Also the saftey feature that are already implemented should stay  the same (if no change in distance and the motor is on it should stop).

iLLiac4, 2020-01-21:
Hi. I have found another interesting project https://www.henrirantanen.fi/2016/04/18/smartdesk-sensor-upgrade/ it is for bekant version but it could be adoptable to this project and uses also ultrasonic sensor.

iLLiac4, 2020-03-19:
Hi. Did you managed to get the ultrasonic sensors?

aenniw, 2020-04-07:
Hi, prepared [POC](https://github.com/aenniw/ARDUINO/tree/feature/HC-SR04), however didn't have time to test it maybe this weekend after I fix the coupler && shaft on my setup...

iLLiac4, 2020-04-07:
Nice.will test.metal shaft coupler works great.

iLLiac4, 2020-04-29:
Hi. I saw that you have made the repo for ultrasonic sensor. Is it already functional and ready to test? Will it display the measured distance on the display? Are you going to use a library for ultrasonic sensor?
Another question are you ditching led screen on v2 because in v2 stl cover there is no more led?

aenniw, 2020-05-03:
1. Yes it should be ready for testing, watchdog may needs tuning as it may false-posive triggers when you move legs under the table... that depends on sensor position...
2. Yes, it's stated in `platformio.ini` dependencies
3. Not sue what do you mean its regarding #76  ?

iLLiac4, 2020-05-03:
Hi. Tnx for the answer. 

I will compile it and test the ultrasonic. Maybee just to avoid the misfiring of ultrasonic sensor maybee it would be possible to make it active only when the button is pressed and if there is no movement for 1 sec then they should stop be active.

iLLiac4, 2020-05-04:
Hi. I have compiled and tested it on relay version. It does not work. The distance sensor is measuring but the motors spins even when error 1 or. 2 occur which will slam the table into the physical endstops. If motor is spinnig and you press any button the motor does not stop like it did in other versions which is security issues.
For now I have tested it not mounted on the table yet but atm it looks like it is not safe yet to mount it on the table. Or. do you think that I should put it on the table with load and test it?
Also as I have noted there is no need to constant doing the measurement since you can easy trig err. with legs.

iLLiac4, 2020-05-05:
Hi. I have tested it on the table. It works but there are still some problems with memory functions. It looks like that it sometimes misses the position and then throws err.1.
It looks like it is not updating the measurement quick enough.
But overall I think it will be better than rotary. It is important where to mount the echo sensor so it does not get disturbed from other things. I have temp. mounted it between legs and works good.

iLLiac4, 2020-05-08:
There seems also to be a problem with led display not going into sleep mode. It looks like it does not go into sleep only when above 1000.

aenniw, 2020-05-08:
Will look into it more this weekend... however:
1. The logic of display wasn't altered, the reason why it doesn't dim is due to sensor, as its values changes even when the table isn't moving... and the display dims only when no change was made for period of time...
2. Minimum stable sample rate of sensor is around 125ms, with anything lower the measurement deviation increases quite a lot and measurements aren't usable at all...
3. Disabling the watchdog after table stops moving isn't good idea, as the sole purpose of watchdog is to detect unintended movement and deadlocks... Timeouts should be tweaked, and will add option to disable it via flag, however it should be enabled by default

Added most of the issues that can be addressed in the PR...

iLLiac4, 2020-05-08:
The problem with updating the measurement is that it misses the stored position and it just overrides it. This happens when lowering the table since it is moving faster than when uplifting.
Do you think that library mentioned above would improve that or. not?

iLLiac4, 2020-05-08:
Do you think it would make sense to somehow combine ultrasonic sensor with rotary encoder to always auto calibrate min and max hight?

iLLiac4, 2020-05-09:
The ultrasonic sensor could be active only if motor relay is active. If not then there is no need to be active. An if there is no input from sensor and the motor is active then err. Is ok.

aenniw, 2020-05-11:
1. library above did it with `delays(500)` so that wont solve it, as some period of time is needed for stabilization...
2.it could be used as end stops that have hardcoded values... however that will drop the option of custom end stop and may not work correctly if table cannot move properly...
3. updated the timeout to be more responsive and now invalid reading are skipped so display will dimm correctly and watchdog wont trigger randomly...

iLLiac4, 2020-05-11:
I have test it and those err. 1 are PITA., I can not set it to work without triggering error.
What can i do to make it more tolerant? Those settings are a little confused.
#ifdef __USENSOR__
    #define WATCHDOG_TIMEOUT        550
#else
    #define WATCHDOG_TIMEOUT        100
#endif
#define WATCHDOG_DEADLOCK_CHANGE    100
#define WATCHDOG_OTHER_CHANGE       100
#ifdef __H_BRIDGE_MOTOR__
    #define WATCHDOG_TOLERANCE      20
#else
    #define WATCHDOG_TOLERANCE      3
#endif

And is it normal that display sometimes blanks during the counting.

bisselke, 2020-05-17:
Currently I am trying to test this feature on a setup with the H-Bridge and only a HC-SR04 sensor, but I get immediately an error 4. Is this because I don't have the rotary encoder, or could there be another problem? Thanks for all the hard work btw! Is there maybe a way to donate something?

aenniw, 2020-05-19:
1.  try to increase `WATCHDOG_DEADLOCK_CHANGE`, however it seems strange that you have issues with `Err-1` as I'm mostly experiencing `Err-2` when accidentally trigered by legs...
2. It should blink only during calibration, when the issue occurs?

iLLiac4, 2020-05-24:
I have set the settings as other user reporting to:
#define WATCHDOG_TIMEOUT 2000
#define WATCHDOG_DEADLOCK_CHANGE 20
#define WATCHDOG_OTHER_CHANGE 100
#define WATCHDOG_TOLERANCE 20

And it seems that it now works. I have mounted the sensor so that I can not trigger it. 
Maybe also it would be nice to migrate the option into platformio.ini the settings for the  offset of the position that will be ignored. For example if i set the hight position to 1000 and the sensors stops at 980 if I press the button again it will go to 1000 even if there is no need and could be ignored. So if there is an option to set like 20 the offset ignored position it would do nothing if I press the button again. But if there would be more like 20 at 970 it will trigger and go to 1000.

aenniw, 2020-05-26:
yeah that would be nice to extract out of `Motor.h` :+1: , also I've updated min step when motor is running so that more changes are propagated to `WD` to reduce `err-1` could you check it out? Thx

iLLiac4, 2020-05-27:
Hi. I have tested it and it works, but I have a feeling that there are now err1 more often then before even with those settings 
#define WATCHDOG_TIMEOUT 2000
#define WATCHDOG_DEADLOCK_CHANGE 20
#define WATCHDOG_OTHER_CHANGE 100
#define WATCHDOG_TOLERANCE 20

Also I do not know maybe just a feeling but there are now more occasions when motor during the drive just stops. Without error or any thing. When you press the button again it continues. This is just like the other user has reported. But I can tell you that this issue is not something that has been introduced lately but it exists from the beginning, if this info helps and is not rely or. hbridge specific.

aenniw, 2020-05-27:
Could you provide logs when these issues occurs if possible? Thx

iLLiac4, 2020-05-27:
Can you please tell me where can i get logs?

aenniw, 2020-05-27:
you need to compile && upload firmware with `-D__DEBUG__` and then via serial `9600` you can collect debug logs, however when you connect arduino to PC disconect it from PSU when testing ...

closed 2020-06-29 by aenniw

ggajews, 2021-12-02:
@iLLiac4 have you solved the Err 1 issue?
Yesterday I replaced the pulley with HC-SR04, and I'm getting a lot of Err 1.
I flashed latest ubridge with default settings.
I noticed that readings from HC-SR04 are not continues. I.e 0610 -> 0640 -> 0660 ... so this might be a reason