aenniw/ARDUINO #70: DC motor - not enough power

TigerClaw33, 2020-01-20:
Hi, yesterday I tested my build but it seems the the DC engine you recommended has not enough power. It went slower and slower and stopped. And if I drive the table down it stucks maybe because its to fast. I dont think my table is to heavy ... I would use some other dc engine low for jalousi but I don't know which one. Or maybe my powersupply is not good enough. I uses a power supply with 12V 8,5A and also one with 12,5A. Maybe not enough

iLLiac4, 2020-01-20:
It has enough amps. Under the load it takes 3-4A. If the motor would use all the power from your PowerSupply then it would burn at around 10A or. even less.  Sometimes the Chinese power supplyies are rated 10A plus and they do not deliver not even 1A.
I have seen that relay version has more power on one of my desk. Also level the desk. If it goes into error when lowering then adopt the code sensitivity and recompile. 
And also have seen bad pwm Bts7960b boards.

TigerClaw33, 2020-02-15:
Legs are straight. Cannot be modified. I could take a different motor from a car but I think it will not really fit to the case and maybe I will need different ratio and I don't have experience in this

aenniw, 2020-05-12:
This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days

closed 2020-06-19 by aenniw

yurazazik, 2021-01-16:
> It has enough amps. Under the load it takes 3-4A. If the motor would use all the power from your PowerSupply then it would burn at around 10A or. even less. Sometimes the Chinese power supplyies are [...]

What do you mean by level the desk?