aenniw/ARDUINO #83: Motors stops very often when going up

bisselke, 2020-05-18:
Hi, I am using the project with the H-bridge and the HC-SR04. Going down is working perfectly but when going up it stops several times when trying to go to the top level. I already tries to adjust the watchdog parameters and the speed, but that doesn't make any differnce other, than that the error doesn't occur so fast. What causes the motor to stop? Is it a security in the motor itself to stop it from overheating, or something else? I already leveled the desk and the legs are straight.
Thanks in advance, and keep up the good work!

bisselke, 2020-05-19:
Could it be possible that the inaccuracy of the HC-SR04 sensor causes it to stop? It measurements during the same table position can differ around 2 to 3 cm.

aenniw, 2020-05-19:
Did the motor stop due to any of the `Err-1` or `Err-2`?

bisselke, 2020-05-20:
With the original WATCHDOG settings (from the HC-SR04 branch) I get a Err-1. 

After adjusting the Watchdog settings I no longer get the Err-1, however the motor still stops so it isn't fixed.  

#define WATCHDOG_TIMEOUT        2000
#define WATCHDOG_DEADLOCK_CHANGE    20
#define WATCHDOG_OTHER_CHANGE       100
#define WATCHDOG_TOLERANCE      20

What can cause the motor to stop? It doesn't looks like the error itself is stopping the motor, but the error is an result when the motor stopped. Weight or power isn't an issue to the motor itself, because when I am leaning on the desk to make it more heavy, it doesn't trigger the motor to stop, it just gets slower.

iLLiac4, 2020-05-24:
I do see see those occasional stops that are much more often when going down. The motor just stops without any error. Then you press the position button again and motor continuous.

As a note this has nothing to do with the ultrasonic sensor since this issue was presented like from the beginning but I was not paying too much attention to it.

bisselke, 2020-05-25:
After using the desk for a week, I can confirm that the sensor is not the issue, because when I put my leg under it for a sec, the motor just continues (because of my watchdog settings). I also experience the stopping problem when going down, although not as much as when the desk is going up. I really don't understand why it stops though...

aenniw, 2020-05-26:
which version of setup you use?
- [ ] relays + encoder
- [ ]  relays + HC-SR04
- [ ] h-bridge + encoder
- [ ] h-bridge + HC-SR04

does the arduino restarts itself when the motor stops?
fixed few issues in `feature/HC-SR04` during weekend & extended logging could you try it out?

`logs` would be great!

bisselke, 2020-05-26:
I am using the h-bridge + HC-SR04. It does not restarts itself. I will first try your latest version with extended logging. If I encounter problems I will post it with a log next time. Thanks in advance!

bisselke, 2020-06-05:
It is working a lot better, although not flawless, but only when I use the adjusted watchdog settings I posted above, otherwise it will still get an error very fast. 
I uploaded 2 log files, one with the original configuration, and one with the less stricted configuration I posted above.

[skarsta_29-05-2020_original_configuration.log](https://github.com/aenniw/ARDUINO/files/4734490/skarsta_29-05-2020_original_configuration.log)
[skarsta_29-05-2020_less_strict_configuration.log](https://github.com/aenniw/ARDUINO/files/4734491/skarsta_29-05-2020_less_strict_configuration.log)

aenniw, 2020-09-15:
The issue should be resolved in via #100 could you check if you still have the problem? Thx

closed 2020-09-15 by aenniw