aenniw/ARDUINO #94: Err-3

r34220, 2020-08-28:
I am hoping you can help. I am getting Err 3 at boot with nanoatmega328-ubridge. I am using the KY-040 Rotary Encoder. If I use nanoatmega328-bridge table will travel up and down but of course the counter doesn't work which I believe is expected.

Can you help?

aenniw, 2020-08-31:
`u` prefixed build are for `ultrasonic` sensor only thus thats the reason for `Err 3`
Regardig the rotary encoder validate that the wiring is ok, check that the encoder is working on simple sketch or flash debug build to microcontroller and reproduce the issue and upload the logs here...

r34220, 2020-08-31:
Looks like the encoder was wired wrong. It was unclear to me what wiring
was based on the drawings.

Thank you for your quick response. I am very excited its all working.

Thanks for your work on this !!

closed 2020-09-01 by aenniw