aenniw/ARDUINO #95: Ultrasonic off option (req)

iLLiac4, 2020-09-02:
Hi.

Is it possible to add an option for ultrasonic to not always do the measuring but only when the motors are on.
After the motors turns of it can stop measure after like 5 sec, And they should turn on after the press of the button.

It is annoying because i always get err.2 because of my legs moving or. vacuum cleaner triggering them. I could move the sensor to other position but it wont help because it has to be perfect spot to do the proper measuring.

TNX

aenniw, 2020-09-06:
That could be done, will look onto it next week...

aenniw, 2020-09-15:
Feature should be available for `ultrasonic sensors` by default with threshold of `5` minutes

closed 2020-09-15 by aenniw