aenniw/ARDUINO #96: Panel cover v2 mesh

micharaze, 2020-09-02:
It looks like the mesh of the panel cover v2 is broken. When I try to slice it with Cura, it fills the gap where the display should be stuck. Also the github preview of the cover shows a hole on the Botton: https://github.com/aenniw/ARDUINO/blob/master/skarsta/models/panel-v2/panel-cover.stl

Here a picture of the sliced cover:
[image]

iLLiac4, 2020-09-02:
it is not broken it is semi transparent.

micharaze, 2020-09-02:
Sorry for the word "broken", I mean it's just not working for me. If you can see in the sliced picture, the gap is completely filled out. I also printed it already and its not transparent. Even the 4 holes are not there. 

What did you used for a slicing software?

iLLiac4, 2020-09-02:
Then take version one.

micharaze, 2020-09-03:
There is definitely something wrong  with the stl. I opened it with different previewers and 3d tools and all have the same issue. Even Fusion 360 shows that the stl is not a full object, there are missing areas. Cura also gives the message that areas are missing.

aenniw, 2020-09-06:
Yep, that model is corrupted I didn't noticed it as [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/releases) fixes them automatically...

Fixed the STLs could you recheck them? [panel-cover.stl](https://github.com/aenniw/ARDUINO/blob/master/skarsta/models/panel-v2/panel-cover.stl) Thx

closed 2020-09-06 by aenniw

micharaze, 2020-09-06:
Looks fine in Cura now, thanks! I will print it tomorrow and check the results.