aenniw/ARDUINO #97: Error 1 : Cant find what is faulty

IluvatarBZH, 2020-09-06:
Hi,
I did the assembly with the first version (h-bridge and Ky-040) some time ago but I had error 1 all the time.
First time I thinking it was a faulty motor (there was hard spot on the gearing) so I change it but still Err1.

After receiving some other spare parts I changed what seemed to me to be the cause of this error (ky-040 and h-bridge) but still have err1.
The motor runs well with the 12V in direct.

I managed only once to have numbers on the display after testing a 3rd ky-040 but after some manipulation off going up and down it always ended with error 1, so i have to restart and after 4-5 time doing this it came back with the 4 blinking dashes (maybe it's important, I did the manipulation without connecting it to the desk)
I feel like the ky-040s are burning, what can lead to that? I have a good +5v on it.
Thank you
(Sorry for my english)

IluvatarBZH, 2020-10-06:
Nobody ? What can it be ? Code or wiring ?

aenniw, 2020-10-06:
Try to adjust `WATCHDOG` timeout values in [configuration.h](https://github.com/aenniw/ARDUINO/blob/master/skarsta/src/configuration.h) 
also could you try to compile with `HALF_STEP` as some of the `KY-040` seems to skip steps...

Could you provide some logs if possible? compile with `__DEBUG__`

Or in worst case compile without `__WATCHDOG__` to dissable internal checks...

yusufiga, 2020-10-10:
I had the same issue. If you test without connecting your motor to the table most likely your motor will not stop in 100ms as there is no counter force like table's weight and watchdog will give you err1. You can adjust the value to some seconds for testing as @aenniw suggested. For test purposes I ended up changing WATCHDOG_TIMEOUT value to 15000 it all worked fine. 

NOTE: don't leave WATCHDOG_TIMEOUT value at 15000 when you connect to the table, it's way too long ;)

closed 2020-11-16 by github-actions[bot]

IluvatarBZH, 2020-11-27:
Hi, i managed to get rid of err1 by changing watchdog setting to this :

#ifndef H_BRIDGE_MOTOR
#define WATCHDOG_TIMEOUT 2000
#define WATCHDOG_DEADLOCK_CHANGE 20
#define WATCHDOG_OTHER_CHANGE 100
#define WATCHDOG_TOLERANCE 20

Now it dont have this error anymore but but the height increment is not done well
It looks like missed step or even sometime while the desk go up the panel numbers retunrs to 0
A the higher postion i was a 4 on the panel so I can't going down with the motor then