aenniw/ARDUINO #106: Error 1 treshold

iLLiac4, 2020-09-16:
Can you move Err 1 into plaformio. I have set those parameters into 

#ifdef __USENSOR__
    #define WATCHDOG_TIMEOUT        2000
#else
    #define WATCHDOG_TIMEOUT        100
#endif
#define WATCHDOG_DEADLOCK_CHANGE    2000
#define WATCHDOG_OTHER_CHANGE       100
#ifndef WATCHDOG_OTHER_SLEEP
    #define WATCHDOG_OTHER_SLEEP    0
#endif
#ifdef __H_BRIDGE_MOTOR__
    #define WATCHDOG_TOLERANCE      5
#else
    #define WATCHDOG_TOLERANCE      20
#endif

And it still throws Err.1 constantly. I can not uplift the desk without getting 5x err.1

aenniw, 2020-09-27:
you can disable the `watchdog` via removing `__WATCHDOG__` from `platformio.ini` see `build_flags`

iLLiac4, 2020-10-05:
TNX it works.

closed 2020-10-05 by iLLiac4