aenniw/ARDUINO #107: Calibration doesn't work

yusufiga, 2020-09-22:
Hello, 

First of all thank you for the project and sharing with  the community. Lately I have assembled the project. Everything works except calibration.

As it's explained on the documentation while "----" blinking, I start lowering the table and press tilde for 3 seconds then "0000" appears and starts blinking on the screen. When I start pressing up motor rotates and moves up but number on the display never doesn't change. From this point on I cannot continue because if I save by using tilde it saves start and end position as 0 and stops moving.

At first I thought KY-040 Rotary Encoder is not working but I doubt that because when I remove the cables and try to move I get Err1. I assume if rotary encoder would be faulty it wouldn't move up and down while it's uncalibrated?

I'm using bridge version. Any idea what could be wrong?

Thanks

closed 2020-09-22 by yusufiga

itaysarig, 2020-12-23:
Do you mind explaining exactly what was your wiring issue? I think i have the exact same issue but can't seem to find my wiring mistake.
@yusufiga

yusufiga, 2020-12-23:
Hi @itaysarig, I was trying to connect wires as it's shown on the image without paying attention to the labels on rotary encoder. 

D2 on Arduino should be connected to CLK on rotary encoder. 
D3 on Arduino should be connected to DT on rotary encoder.

itaysarig, 2020-12-24:
Thanks for the help tried it out and it fixed it.