aenniw/ARDUINO #110: Wrong direction

tradzieda, 2020-10-04:
Hi,

i have the problem that the motor moves in the wrong direction. If I look up at the shaft, it turns to the left when I press the button up and the count goes up. But the table goes down.

I checked the wiring but can't find the problem.

 It seems to me that the rotary encoder (ky-040) is programmed the wrong way round.

tradzieda, 2020-10-04:
https://youtu.be/SLC-cCDy5Zs

my key sorting is
set, 1, 2, 3, Up, Down

iLLiac4, 2020-10-04:
Revers motor position.

tradzieda, 2020-10-04:
> Revers motor position.

Is there a configuration for this in the program code?

iLLiac4, 2020-10-04:
Hmm. I would phisically switck wires going to the motor.

tradzieda, 2020-10-04:
> Hmm. I would phisically switck wires going to the motor.

When I do this, the motor turns in the other direction, but the display is wrong. When I press the button up, the display goes down. Same the other way round.

So the error is between key and display.

iLLiac4, 2020-10-05:
https://github.com/aenniw/ARDUINO/issues/75

aenniw, 2020-10-05:
If you have relay version just upload one of these based on your sensor...
```
nanoatmega328-irelays
nanoatmega328-iurelays
```

tradzieda, 2020-10-05:
> If you have relay version just upload one of these based on your sensor...
> 
> ```
> nanoatmega328-irelays
> nanoatmega328-iurelays
> ```

thanks, but i have the H-Bridge Version

> #75
i will look, thanks

tradzieda, 2020-10-05:
changing pin from Sensor dosnt work for me

```
#define SENSOR_PIN0 2
#define SENSOR_PIN1 3
```

aenniw, 2020-10-05:
try to update 

```c
    if (state == CCW) {
        analogWrite(l_pwm, 0);
        analogWrite(r_pwm, speed);
    } else if (state == CW) {
        analogWrite(r_pwm, 0);
        analogWrite(l_pwm, speed);
    }
```
to
```c
    if (state == CCW) {
        analogWrite(r_pwm, 0);
        analogWrite(l_pwm, speed);
    } else if (state == CW) {
        analogWrite(l_pwm, 0);
        analogWrite(r_pwm, speed);
    }
```
in `MotorBridge.cpp`

tradzieda, 2020-10-05:
@aenniw 
it works together with changing the sensor

```
#define SENSOR_PIN0 3
#define SENSOR_PIN1 2
```

thx for help. Perhaps you could provide an option for this in configuration.h

aenniw, 2020-10-05:
updated `configuration.h` so that polarity can be reversed in, #111  could you check if it works? Thx

closed 2020-10-06 by aenniw

radze90, 2020-10-17:
I cant check, but thanks for add this feature. my motor is somehow too weak ...

PR #111 (2020-10-06): Add motor reverse mode
- add option to configure reverse motor mode
- update header guards to pragma

closes #110