aenniw/ARDUINO #114: BRIDGE or RELAYS?

Oeil2L, 2020-10-16:
Hello

Sorry for this questions on issues zone but didn't find a way to clarify some points.

I'm interested by the project (more for the fun/learning than the functions itself ;)), what's the advantage/cons between BRIDGE or RELAYS version? (precision, security, ...)

Because I have already a FIXA screwdriver button broken, the motor still works, can I use it with one of this 2 versions? both?

PS: stupid question, changing the Arduino nano by an other, 
would it be possible to implement wifi? (Geek stuff ;))

Thanks for your answers.
Anyway, great project with a lot of details.

aenniw, 2020-11-10:
| RELAYs | BRIDGE |
|:-----------:|:------------:|
| + cost less (common parts)| - higher cost (not that common part) |
| - only on/off states | + adjustable speed of motor |
| + no throughput degradation | - possible throughput degradation |

Regarding the FIXA motor:
- you can use it however you will need really beefy PSU for it.... ( i would recommend using other DC motor for that...)
- you will need to remodel mounts for the FIXA motor as current models are designed only for 1 type po motor mounting

You can use it with other microcontrollers however you need to update some of the code based on your chip like:
- interrupt handling
- ADC resolution

Oeil2L, 2020-11-10:
Thanks a lot for you answers aenniw.

more clear for me, I will order tomorrow the right BOM.

For the microcontrollers, I thought about nodeMCU in replacement of the Arduino Nano, but it work on 3,3V and not at 5V like nano.

++

aenniw, 2020-11-10:
in #61 nodeMCU ADC was mentioned so for more info try to reach out @p0thi ...

closed 2020-11-11 by aenniw