aenniw/ARDUINO #116: vector: no such file or directory

sxaris87, 2020-11-25:
hello i try to upload the code to ardruino nano and i dont know the way.
1st try was with this error (vector: no such file or directory)
could you please tell me step by step?

sxaris87, 2020-11-27:
Sorry but i need someone to lead me the way to programming the arduino nano.
i download PlatformIO and i m trying to figure it out but its difficult to me.

IluvatarBZH, 2020-12-02:
Download vscode. 
Install it and run it. 
From the extension menu, search for platformio and install it. 
Reload when it asks you, and create a new project with it. Name it whatever, at board select Arduino nano and let it finish.
Close it. 
Navigate to the project folder and copy from the GitHub page the following: 
platformio.ini 
SRC folder
lib folder
Paste them in the project folder, overwrite where necessary. 
Then open vscode, open the project, compile it to see if there are any errors and then upload it to a nano.

sxaris87, 2020-12-02:
Thank you for your answer. Looks helpful. I will do it and when it was ready and see if runs correctly and then a close this case and I will post my try. 
Thank you

sxaris87, 2021-01-06:
@IluvatarBZH  i have done the process you told me but now i cant upload to arduino 
got this error:
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x3e

can i clean the arduino? how i reset it like new?

closed 2021-01-12 by github-actions[bot]