aenniw/ARDUINO #120: Motor Pully can't seem to hold the grip on shaft

itaysarig, 2020-12-24:
Tried printing the  part using PLA 100% infill a couple of times and PLA couldn't hold at all.
Tried to print use PETG and 100% infill and still it seems the screw that's holding the shaft doesn't seem to have enough  pressure on it.
When facing actual desk load it gives.
Was hoping to know maybe what type of screw I should use or did anyone faced the same issue.

iLLiac4, 2020-12-24:
Yes printed one is not good. I have used metal one. I think it is listed in readme.

itaysarig, 2020-12-24:
> Yes printed one is not good. I have used metal one. I think it is listed in readme.

mind linking it? i can't find it

iLLiac4, 2020-12-24:
https://github.com/aenniw/ARDUINO/issues/24

itaysarig, 2020-12-24:
> #24

Thank but the link to ebay is a dead link, do you maybe have a live link?

closed 2020-12-24 by itaysarig