aenniw/ARDUINO #121: cant upload to arduino avrdude: stk500_recv(): programmer is not responding

sxaris87, 2021-01-06:
@IluvatarBZH hello i do it like  you told me and got many errors (12) at built and failed.
 at  the upload got other error (avrdude stk500_recv() programmer is not responding)
need more help please. i have all the parts and i have done all the wiring and boom (nothing ) many errors and cant upload.

Elmardus, 2021-01-11:
In the platformio.ini try setting
```board = nanoatmega328```
to 
```board = nanoatmega328new```

There are two different bootloader versions of the Arduino Nano, which are not compatible.

sxaris87, 2021-01-12:
Lets take the steps from the beginning.
1) do the wiring for H-bridge version like this https://github.com/aenniw/ARDUINO/blob/feature/opto-skarsta/skarsta/img/wiring-h-bridge.png
2) download Microsoft Visual Studio
3) inside VS at extensions search and download PlatoformIO IDE 
4) home - quick access - new project - pick name and board (Nano ATmega328 or Nano ATmega328 new) - finish 
5) find the project (reveal in file explorer)  and close platformio
6) download the code from here https://github.com/aenniw/ARDUINO/tree/master
7) go to skarsta folder and copy (lib, src and platformio.ini) and replace at step 5) at your project 
8) open platformio again and run build to see if any errros, if not, flash to your arduino.

is this procedure wrong? something is missing? why i cant do this? need to download something  more?
someone how make it work please make step by step instructions for stupids like me and help us.

Elmardus, 2021-01-12:
Procedure looks right to me. Do you have an unconnected arduino to test the upload with?

sxaris87, 2021-01-12:
I order a new one to see if something went wrong with this. I think I have it at the next days

sxaris87, 2021-01-15:
new adruino nano just arrinve and make this steps like i post before and got 6 errors at built

nanoatmega328-relays    FAILED    00:00:01.024
nanoatmega328-urelays   FAILED    00:00:01.023
nanoatmega328-irelays   FAILED    00:00:01.014
nanoatmega328-iurelays  FAILED    00:00:01.010
nanoatmega328-bridge    FAILED    00:00:01.004
nanoatmega328-ubridge   FAILED    00:00:01.010

any ideas?

iLLiac4, 2021-01-15:
You have to comment out things that you do not use.

platformio]
default_envs =
 ;  nanoatmega328-relays
 ;   nanoatmega328-urelays
 ;   nanoatmega328-irelays
 ;   nanoatmega328-iurelays
 ;   nanoatmega328-bridge
  nanoatmega328-ubridge

[common]
watchdog_sleep = 60
debug_build_flags =
    -D__DEBUG__
build_flags =
    -Os -std=c++11 -Wall -Wextra
;    -D__WATCHDOG__
    -D__EEPROM__

[env]
platform = atmelavr
board = nanoatmega328
framework = arduino
src_filter = +<main.cpp>
lib_extra_dirs = ../lib
lib_deps =
    https://github.com/Martinsos/arduino-lib-hc-sr04.git#1.1.0
    ArduinoSTL@1.1.0
    elapsedMillis@1.0.6
    https://github.com/brianlow/Rotary.git#master
    https://github.com/JCWentzel/PolymorphicButtons.git#1.0.3

[env:nanoatmega328-relays]
debug_build_flags = ${env:nanoatmega328-relays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}

[env:nanoatmega328-urelays]
debug_build_flags = ${env:nanoatmega328-urelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
              -D__USENSOR__

[env:nanoatmega328-irelays]
debug_build_flags = ${env:nanoatmega328-irelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -D__INVERSE_RELAYS__

[env:nanoatmega328-iurelays]
debug_build_flags = ${env:nanoatmega328-iurelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
              -D__INVERSE_RELAYS__
              -D__USENSOR__

[env:nanoatmega328-bridge]
debug_build_flags = ${env:nanoatmega328-bridge.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -D__H_BRIDGE_MOTOR__

[env:nanoatmega328-ubridge]
debug_build_flags = ${env:nanoatmega328-ubridge.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
              -D__H_BRIDGE_MOTOR__
              -D__USENSOR__

[env:nanoatmega328-ptester]
build_flags = ${common.build_flags}
src_filter = +<ptester.cpp>

sxaris87, 2021-01-15:
with common words please?

sxaris87, 2021-01-15:
@iLLiac4 do you know to make step by step the uploading? 
from creating the project until the files that need it to cope paste to make the adruino work when i connected to the hole project

sxaris87, 2021-01-15:
i think its so simple for someone who knows this things but i dont.
think that to upload something to arduino its easy.

sxaris87, 2021-01-15:
@iLLiac4  have to connect everything and then to plug in arduino to pc for the uploading?

Elmardus, 2021-01-15:
In the platformio.ini file you can have only one of the default_envs enabled, so if you wish to use the ubridge mode (bridge driver with ultrasonic sensor, then change platformio.ini to:
```
default_envs =
; nanoatmega328-relays
; nanoatmega328-urelays
; nanoatmega328-irelays
; nanoatmega328-iurelays
; nanoatmega328-bridge
nanoatmega328-ubridge
```
Alternatively, you can also click on 'Default' on the bottom blue bar. Then a popup shows where you can select the version you wish to build:
[image]
This way you do not have to comment out any of the default_envs in ``platformio.ini``

Except for the USB cable you don't have to have anything connected to the Arduino to be able to upload.

sxaris87, 2021-01-15:
i just make that change and now this

> Executing task in folder test13: C:\Users\Sxaris\.platformio\penv\Scripts\pio.exe run <

Error: Invalid 'C:\Users\Sxaris\Documents\PlatformIO\Projects\test13\platformio.ini' (project configuration file): 'While reading from 'C:\\Users\\Sxaris\\Documents\\PlatformIO\\Projects\\test13\\platformio.ini' [line 29]: option 'https' in section 'env' already exists'
The terminal process "C:\Users\Sxaris\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

sxaris87, 2021-01-15:
thank you for helping me guys, what is this error now?

Elmardus, 2021-01-15:
What do you have on line 29 on your platformio.ini file? That is where the error points to. If not clear please post the complete file.

sxaris87, 2021-01-15:
[platformio]
default_envs =
; nanoatmega328-relays
; nanoatmega328-urelays
; nanoatmega328-irelays
; nanoatmega328-iurelays
; nanoatmega328-bridge
nanoatmega328-ubridge

[common]
watchdog_sleep = 60
debug_build_flags =
-D__DEBUG__
build_flags =
-Os -std=c++11 -Wall -Wextra
-D__WATCHDOG__
-D__EEPROM__

[env]
platform = atmelavr
board = nanoatmega328
framework = arduino
src_filter = +<main.cpp>
lib_extra_dirs = ../lib
lib_deps =
https://github.com/Martinsos/arduino-lib-hc-sr04.git#1.1.0
ArduinoSTL@1.1.0
elapsedMillis@1.0.6
https://github.com/brianlow/Rotary.git#master
https://github.com/JCWentzel/PolymorphicButtons.git#1.0.3

[env:nanoatmega328-relays]
debug_build_flags = ${env:nanoatmega328-relays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}

[env:nanoatmega328-urelays]
debug_build_flags = ${env:nanoatmega328-urelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
-DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
-D__USENSOR__

[env:nanoatmega328-irelays]
debug_build_flags = ${env:nanoatmega328-irelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
-D__INVERSE_RELAYS__

[env:nanoatmega328-iurelays]
debug_build_flags = ${env:nanoatmega328-iurelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
-DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
-D__INVERSE_RELAYS__
-D__USENSOR__

[env:nanoatmega328-bridge]
debug_build_flags = ${env:nanoatmega328-bridge.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
-D__H_BRIDGE_MOTOR__

[env:nanoatmega328-ubridge]
debug_build_flags = ${env:nanoatmega328-ubridge.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
-DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
-D__H_BRIDGE_MOTOR__
-D__USENSOR__

[env:nanoatmega328-ptester]
build_flags = ${common.build_flags}
src_filter = +<ptester.cpp>

sxaris87, 2021-01-15:
line 29 is this : https://github.com/brianlow/Rotary.git#master

sxaris87, 2021-01-15:
> 
> 
> What do you have on line 29 on your platformio.ini file? That is where the error points to. If not clear please post the complete file.

line 29: https://github.com/brianlow/Rotary.git#master

sxaris87, 2021-01-15:
its was my mistake this but still got 2 warnings

In file included from src\main.cpp:3:0:
lib\display/Display.h:4:21: fatal error: Service.h: No such file or directory

*****************************************************************
* Looking for Service.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Service.h"
* Web  > https://platformio.org/lib/search?query=header:Service.h
*
*****************************************************************

compilation terminated.
*** [.pio\build\nanoatmega328-ubridge\src\main.cpp.o] Error 1
.pio\libdeps\nanoatmega328-ubridge\ArduinoSTL\src\ArduinoSTL.cpp:54:42: warning: unused parameter 'f' [-Wunused-parameter]
 static int arduino_putchar(char c, FILE* f) {
                                          ^
.pio\libdeps\nanoatmega328-ubridge\ArduinoSTL\src\ArduinoSTL.cpp:65:34: warning: unused parameter 'f' [-Wunused-parameter]
 static int arduino_getchar(FILE *f) {
                                  ^
======================================================================= [FAILED] Took 1.18 seconds =======================================================================

Environment            Status    Duration
---------------------  --------  ------------
nanoatmega328-ubridge  FAILED    00:00:01.176
================================================================== 1 failed, 0 succeeded in 00:00:01.176 ==================================================================The terminal process "C:\Users\Sxaris\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Elmardus, 2021-01-15:
Download the whole github repo, and not just the Skarsta subfolder.  There is a folder with Service.h and display.h in the top level of this repository.

sxaris87, 2021-01-15:
i download form here https://github.com/aenniw/ARDUINO
Code -> download ZIP
1) i create new project at platformIO, 
2) picking a name (test17), 
3) selecting a board arduino nano atmega328(new bootloader),
4) framework: Arduino
5) clicking finish and i close it
6) going to direction of the project C:\Users\Sxaris\Documents\PlatformIO\Projects\test17 
7) from the Code i download and from skarsta folder i select the folder lib & src & platformio.ini and replace them to test17 folder
8) then i open VS and press build

ps: i also made the change you told me
default_envs =
    ;nanoatmega328-relays
    ;nanoatmega328-urelays
    ;nanoatmega328-irelays
    ;nanoatmega328-iurelays
    ;nanoatmega328-bridge
    nanoatmega328-ubridge

what i do wrong?

sxaris87, 2021-01-15:
**now i took also the lib folder outside skarsta folder and now not failed**

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  36.4% (used 746 bytes from 2048 bytes)
Flash: [====      ]  36.9% (used 11338 bytes from 30720 bytes)
==================================================== [SUCCESS] Took 4.74 seconds ====================================================

Environment            Status    Duration
---------------------  --------  ------------
nanoatmega328-ubridge  SUCCESS   00:00:04.738
==================================================== 1 succeeded in 00:00:04.738 ==================================================== 

Terminal will be reused by tasks, press any key to close it.

**got 20 problems, is that ok?**

sxaris87, 2021-01-15:
default_envs =
;nanoatmega328-relays
;nanoatmega328-urelays
;nanoatmega328-irelays
;nanoatmega328-iurelays
;nanoatmega328-bridge
nanoatmega328-ubridge

**from this i need only ubridge? bridge not need it?**

yurazazik, 2021-01-15:
> default_envs =
> ;nanoatmega328-relays
> ;nanoatmega328-urelays
> ;nanoatmega328-irelays
> ;nanoatmega328-iurelays
> ;nanoatmega328-bridge
> nanoatmega328-ubridge
> 
> **from this i need only ubridge? bridge not need it?**

You only need to choose one, if you collected the circuit on the encoder then choose the bridge, if on the sonar choose the u-bridge

sxaris87, 2021-01-15:
ok and the 20 problems appears due to arduino it now connected to the whole project? it ok to try to upload it to arduino now?

yurazazik, 2021-01-15:
Show what the Vstudio gives.

yurazazik, 2021-01-15:
Did you do full scheme or only connect arduino?

sxaris87, 2021-01-15:
only arduino for now

sxaris87, 2021-01-15:
ok done with upload.
now the 7 segment display show me Err 3

sxaris87, 2021-01-15:
i will try to figure it out or i will ask for help again. 
Thank you all for answering so quickly.

closed 2021-01-15 by sxaris87

yurazazik, 2021-01-15:
> ok done with upload.
> now the 7 segment display show me Err 3

You need check all connections.