aenniw/ARDUINO #123: How much weight does your table lift?

yurazazik, 2021-01-15:
How much equipment can your table lift? In my case, it copes well with the IMac 27 inches weighing 10 kg and even if the table moves down. When the table moves up the motor sprays well somewhere until the middle of the way, and then pain and suffering begin, and when it comes to the top it becomes a tired turtle barely making it to the top. Sometimes his strength is not enough to reach the maximum point and he stops before reaching a couple of centimeters. If I still want to put a second monitor, then I can forget about comfortable work, the trip up looks even more sad. My build setup: I used a bridge circuit with sonar and a 12V 17A computer power supply. I want to try another relay circuit, a miracle may happen :).

yurazazik, 2021-01-16:
Guys give me feedback please.

sxaris87, 2021-01-26:
According to the characteristics of the motor we have? with power supply 12V 20A can handle 90kg.cm 
[image]
but for me to handle 20A you need to change your cables to thicker, 2-2,5mm. 
I don’t recommend to do that. 
I buy an other power supply 12V 10A because the other that I told you was not good.
[image]

yurazazik, 2021-01-26:
I'm waiting for results your report.

yurazazik, 2021-01-26:
I'm waiting for results your report. When i measured Amperes of my motor it was 3-4A in maximum mode.

sxaris87, 2021-01-26:
In relay version that you have your motor works at max rpm? With load max rpm it’s 120? Always according to the motors characteristics.

yurazazik, 2021-01-26:
> In relay version that you have your motor works at max rpm? With load max rpm it’s 120? Always according to the motors characteristics.

I dont know

sxaris87, 2021-01-26:
Can you take a video from sitting position to standing?

yurazazik, 2021-01-26:
> Can you take a video from sitting position to standing?

Yes

yurazazik, 2021-01-26:
https://youtu.be/gANrgU92JSM

yurazazik, 2021-01-26:
did you watch?

sxaris87, 2021-01-26:
Yes.

yurazazik, 2021-01-26:
> Yes.

and what do you think?

sxaris87, 2021-01-26:
You use Aluminum Alloy Drive to connect with desk?

sxaris87, 2021-01-26:
I m thinking about and the only think is the deference in versions relay / h-bridge

yurazazik, 2021-01-26:
I don't understand, what is problem with your power supply? Was It stop to work?

sxaris87, 2021-01-26:
I don’t know really. I bought this that I saw you before. 10A to work with no pressure the power supply. The metallic power supply’s is better than the plastic that I have before.

yurazazik, 2021-01-26:
> I don’t know really. I bought this that I saw you before. 10A to work with no pressure the power supply. The metallic power supply’s is better than the plastic that I have before.

Of course metallic power supply is better

yurazazik, 2021-02-11:
One of these days another similar motor came from China and, what is most surprising, the table began to work much better and more confidently!!! One of these days another similar motor came from China and, what is most surprising, the table began to work much better and more confidently !!! Now the table raises quite calmly two heavy monitors and everything on the table !!!

closed 2021-03-20 by github-actions[bot]