aenniw/ARDUINO #124: 7 segment display error code

sxaris87, 2021-01-15:


sxaris87, 2021-01-15:
hello again, i m done with the uploading to arduino and when i turn it on at 7 segment display show me "Err 3" 
what is the problem with wiring?
any ideas?

yurazazik, 2021-01-15:
Is your circuit already assembled or are you still experimenting with only the arduino board?

sxaris87, 2021-01-16:
no i made all the connection and when i power it up got "Er 3" but i do it like the wiring diagram. 
what its supposed to be wrong? 
what is the requirements to get this code error?

yurazazik, 2021-01-16:
What version and sensor did you use?

sxaris87, 2021-01-16:
h-bridge version. ubridge sensor

yurazazik, 2021-01-16:
Try to disconnect sensor and switch on table.

sxaris87, 2021-01-16:
mmm how to do this?

yurazazik, 2021-01-16:
Show the photo of your connections to 7 segment display.

yurazazik, 2021-01-16:
sorry to your sensor

sxaris87, 2021-01-16:
i disconnect ubridge and i connect bridge and now blinking this to display ----

sxaris87, 2021-01-16:
i think i did it with your help

sxaris87, 2021-01-16:
i connect to my desk tommorow because with no load get error 2

sxaris87, 2021-01-16:
thank you very much for your help @yurazazik

closed 2021-01-16 by sxaris87

yurazazik, 2021-01-16:
could you take a photo of your assembled circuit?

sxaris87, 2021-01-16:
yes but not now because its a draft and i have to sleep, i have to go to work in 3 hours :P