aenniw/ARDUINO #125: Error 1 at 7 segment display

sxaris87, 2021-01-17:
Hello once again. after hour if wiring i connect the project to table and Error 1 appears.
i dont think that the motor cant uplift my table, i have only monitor and keybord up there.
maybe have to change something at the programm?
for last time i need help please

yurazazik, 2021-01-18:
You need to setup the watchdog in configuration.h file. It is depend from your sensor.

sxaris87, 2021-01-18:
**i have this settings and still Error 1**

[platformio]
default_envs =
    ;nanoatmega328-relays
    ;nanoatmega328-urelays
    ;nanoatmega328-irelays
    ;nanoatmega328-iurelays
    nanoatmega328-bridge
    ;nanoatmega328-ubridge

[common]
watchdog_sleep = 5000
debug_build_flags =
    -D__DEBUG__
build_flags =
    -Os -std=c++11 -Wall -Wextra
    -D__WATCHDOG__
    -D__EEPROM__

**and at configuration.h this**

#ifdef __USENSOR__
    #define WATCHDOG_TIMEOUT        5000
#else
    #define WATCHDOG_TIMEOUT        100
#endif
#define WATCHDOG_DEADLOCK_CHANGE    20
#define WATCHDOG_OTHER_CHANGE       100
#ifndef WATCHDOG_OTHER_SLEEP
    #define WATCHDOG_OTHER_SLEEP    0
#endif
#ifdef __H_BRIDGE_MOTOR__
    #define WATCHDOG_TOLERANCE      5
#else
    #define WATCHDOG_TOLERANCE      20
#endif

#define STOP_POS_DIFF       1   // Offset distance for preset end stop
#define MINIMUM_POS_CHANGE  8   // Minimum distance for presets to be invoked

#define R_EN    7
#define L_EN    8
#define R_PWM   9
#define L_PWM   10

#define REVERSE_POLARITY   false
#define DIRECTION_RELAY     6
#define POWER_RELAY         7

sxaris87, 2021-01-18:
maybe have to change something at motorbridge.h? motor.h?

sxaris87, 2021-01-18:
i disable watchdog and without load gives no errors.
when i connect it to my table cant move it up or down. 
anyone please?

yurazazik, 2021-01-18:
What kind motor and motor-pulley did you use? My opinion you need to buy aluminium motor-pulley because it work perfect.

yurazazik, 2021-01-18:
And in file MotorBridge.h try to do this::

#define MIN_SPEED 255
#define MAX_SPEED 255
#define SPEED_STEP_DURATION 0

sxaris87, 2021-01-18:
This motor pulley https://www.ebay.com/itm/264292301651

and this motor 
https://www.ebay.com/itm/233675733031

I buy them all that suggested.

yurazazik, 2021-01-18:
> This motor pulley https://www.ebay.com/itm/264292301651
> 
> and this motor
> https://www.ebay.com/itm/233675733031
> 
> I buy them all that suggested.

It is OK. What kind Power Supply?

sxaris87, 2021-01-18:
This power supply https://www.ebay.com/itm/274067202653

yurazazik, 2021-01-18:
> This power supply https://www.ebay.com/itm/274067202653

Here I'm not sure if it's a good solution, is it possible to try an old power supply from a computer?

sxaris87, 2021-01-18:
with what characteristics to see if i have? 
this was a recommended power suply i think it was in the list

sxaris87, 2021-01-18:
> 
> 
> And in file MotorBridge.h try to do this::
> 
> #define MIN_SPEED 255
> #define MAX_SPEED 255
> #define SPEED_STEP_DURATION 0

the same issue. cant lift it up or down

yurazazik, 2021-01-18:
> with what characteristics to see if i have?
> this was a recommended power suply i think it was in the list

There was also a printed motor-pulley on the list, but unfortunately it doesn't work))

sxaris87, 2021-01-18:
> 
> 
> > with what characteristics to see if i have?
> > this was a recommended power suply i think it was in the list
> 
> There was also a printed motor-pulley on the list, but unfortunately it doesn't work))

yes i saw it but seems not durable to my eyes

sxaris87, 2021-01-18:
what power supply you use?

yurazazik, 2021-01-18:
What is the weight on the table now? Or is it empty?

yurazazik, 2021-01-18:
> what power supply you use?

Im using computer power supply

yurazazik, 2021-01-18:
Have you connected your sensor?

sxaris87, 2021-01-18:
sensor is this? bridge?

[platformio]
default_envs =
    ;nanoatmega328-relays
    ;nanoatmega328-urelays
    ;nanoatmega328-irelays
    ;nanoatmega328-iurelays
    nanoatmega328-bridge
    ;nanoatmega328-ubridge

sxaris87, 2021-01-18:
> 
> 
> What is the weight on the table now? Or is it empty?

only a monitor and a keyboard

yurazazik, 2021-01-18:
> > What is the weight on the table now? Or is it empty?
> 
> only a monitor and a keyboard

Did you try emty table?

yurazazik, 2021-01-18:
> sensor is this? bridge?
> 
> [platformio]
> default_envs =
> ;nanoatmega328-relays
> ;nanoatmega328-urelays
> ;nanoatmega328-irelays
> ;nanoatmega328-iurelays
> nanoatmega328-bridge
> ;nanoatmega328-ubridge

I mean encoder. Did you connect encoder?

sxaris87, 2021-01-18:
rotary encoder you mean?
https://www.ebay.com/itm/1PCS-KY-040-Rotary-Encoder-Module-Brick-Sensor-Development-For-Arduino/381375070024?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

yurazazik, 2021-01-18:
> https://www.ebay.com/itm/1PCS-KY-040-Rotary-Encoder-Module-Brick-Sensor-Development-For-Arduino/381375070024?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Yes, did you put on GT2 160mm belt?

sxaris87, 2021-01-18:
> 
> 
> > > What is the weight on the table now? Or is it empty?
> > 
> > 
> > only a monitor and a keyboard
> 
> Did you try emty table?

no. its not more 7 kg.

sxaris87, 2021-01-18:
> 
> 
> > https://www.ebay.com/itm/1PCS-KY-040-Rotary-Encoder-Module-Brick-Sensor-Development-For-Arduino/381375070024?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
> 
> Yes, did you put on GT2 160mm belt?

yes

yurazazik, 2021-01-18:
> > > > What is the weight on the table now? Or is it empty?
> > > 
> > > 
> > > only a monitor and a keyboard
> > 
> > 
> > Did you try emty table?
> 
> no. its not more 7 kg.

Try empty table!

sxaris87, 2021-01-18:
> 
> 
> > > > > What is the weight on the table now? Or is it empty?
> > > > 
> > > > 
> > > > only a monitor and a keyboard
> > > 
> > > 
> > > Did you try emty table?
> > 
> > 
> > no. its not more 7 kg.
> 
> Try empty table!

same. cant ever lower the table

yurazazik, 2021-01-18:
Does the motor work if you do not connect it to the table?

sxaris87, 2021-01-18:
yes, with watchdog off because Err 1  if watchdog on

sxaris87, 2021-01-18:
its supposed thsi project made for skarsta model and finally cant lift it?
now i have to buy an other power supply? 12V with 10-15A?
or its something else?

yurazazik, 2021-01-18:
> its supposed thsi project made for skarsta model and finally cant lift it?
> now i have to buy an other power supply? 12V with 10-15A?
> or its something else?

I dot't know exactly. I just try to help you)

sxaris87, 2021-01-18:
> 
> 
> > its supposed thsi project made for skarsta model and finally cant lift it?
> > now i have to buy an other power supply? 12V with 10-15A?
> > or its something else?
> 
> I dot't know exactly. I just try to help you)

and thank you for that.
your desk how many kg is?

yurazazik, 2021-01-18:
> > > its supposed thsi project made for skarsta model and finally cant lift it?
> > > now i have to buy an other power supply? 12V with 10-15A?
> > > or its something else?
> > 
> > 
> > I dot't know exactly. I just try to help you)
> 
> and thank you for that.
> your desk how many kg is?

About 12 kg

sxaris87, 2021-01-18:
damn. i had an other power supply  12v 3A and now it can move it up and down. omg the power supply from ebay is a trush. 
i will enable again watchdog to see if Error 1.
and i ll buy a bigger power supply.

sxaris87, 2021-01-18:
thank you so much for your help @yurazazik

yurazazik, 2021-01-18:
> thank you so much for your help @yurazazik

You are welcome)

sxaris87, 2021-01-18:
question for power supply:
my computer is next to my desk, can i power it up with a molex cable?

yurazazik, 2021-01-18:
> question for power supply:
> my computer is next to my desk, can i power it up with a molex cable?

Do you want your computer and desk to work from one power supply unit?

sxaris87, 2021-01-18:
yes its 750watt plat. its working. but i cant set 1,2,3 memmories and display flashing this ( - - - - )fast

sxaris87, 2021-01-18:
watchdog also not working

yurazazik, 2021-01-18:
> yes its 750watt plat. its working. but i cant set 1,2,3 memmories and display flashing this ( - - - - )fast

While display is flashing with dashes, set table to lowest position
Press "tilde" button for 3s or until message '-rst' shows to confirm lowest position and afterwards set table to highest position
Press "tilde" button for 3s or until message '-rst' shows once more to confirm max position
Afterwards you can use table with min/max soft. limits and custom positions
To set new position just press one of 1, 2, 3 buttons until display shows "set" message
To "goto" preset position just press 1, 2, 3 buttons
To reset min/max bounds and custom positions, press "tilde" once more as previously it was calibrated

sxaris87, 2021-01-19:
Still can’t program the memory or high low position maybe something with wiring because when I test it before it works. I will see it tomorrow. Thnx again.

sxaris87, 2021-01-19:
Also I don’t understand why watchdog not working

yurazazik, 2021-01-19:
#define REVERSE_POLARITY   true
what do you see in REVERSE_POLARITY? true or false?

sxaris87, 2021-01-19:
false

yurazazik, 2021-01-19:
try true

yurazazik, 2021-01-19:
> false

Do you have some news?

sxaris87, 2021-01-19:
I m out of home all day, I will try it tomorrow and I will post again. 
your watchdog is working good?

yurazazik, 2021-01-19:
> I m out of home all day, I will try it tomorrow and I will post again.
> your watchdog is working good?

Yes, but i'm using relay version with uSensor. Its work really good. Next step i will buy second motor and i think close a question with weight of my table.

sxaris87, 2021-01-19:
The following is the sernsors?
;nanoatmega328-relays
;nanoatmega328-urelays
;nanoatmega328-irelays
;nanoatmega328-iurelays
nanoatmega328-bridge
;nanoatmega328-ubridge

Could you copy paste your platformio.ini

yurazazik, 2021-01-19:
;nanoatmega328-urelays

yurazazik, 2021-01-19:
[platformio]
default_envs =
    nanoatmega328-relays
    nanoatmega328-urelays
    nanoatmega328-irelays
    nanoatmega328-iurelays
    nanoatmega328-bridge
    nanoatmega328-ubridge

[common]
watchdog_sleep = 300
debug_build_flags =
    -D__DEBUG__
build_flags =
    -Os -std=c++11 -Wall -Wextra
    -D__WATCHDOG__
    -D__EEPROM__

[env]
platform = atmelavr
board = nanoatmega328
framework = arduino
src_filter = +<main.cpp>
lib_extra_dirs = ../lib
lib_deps =
    https://github.com/Martinsos/arduino-lib-hc-sr04.git#1.1.0
    ArduinoSTL@1.1.0
    elapsedMillis@1.0.6
    https://github.com/brianlow/Rotary.git#master
    https://github.com/JCWentzel/PolymorphicButtons.git#1.0.3

[env:nanoatmega328-relays]
debug_build_flags = ${env:nanoatmega328-relays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}

[env:nanoatmega328-urelays]
debug_build_flags = ${env:nanoatmega328-urelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
              -D__USENSOR__

[env:nanoatmega328-irelays]
debug_build_flags = ${env:nanoatmega328-irelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -D__INVERSE_RELAYS__

[env:nanoatmega328-iurelays]
debug_build_flags = ${env:nanoatmega328-iurelays.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
              -D__INVERSE_RELAYS__
              -D__USENSOR__

[env:nanoatmega328-bridge]
debug_build_flags = ${env:nanoatmega328-bridge.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -D__H_BRIDGE_MOTOR__

[env:nanoatmega328-ubridge]
debug_build_flags = ${env:nanoatmega328-ubridge.build_flags} ${common.debug_build_flags}
build_flags = ${common.build_flags}
              -DWATCHDOG_OTHER_SLEEP=${common.watchdog_sleep}
              -D__H_BRIDGE_MOTOR__
              -D__USENSOR__

[env:nanoatmega328-ptester]
build_flags = ${common.build_flags}
src_filter = +<ptester.cpp>

sxaris87, 2021-01-19:
I use only 
nanoatmega328-bridge

and at configuration.h this

#ifdef USENSOR
#define WATCHDOG_TIMEOUT 2000
#else
#define WATCHDOG_TIMEOUT 100
#endif
#define WATCHDOG_DEADLOCK_CHANGE 2000
#define WATCHDOG_OTHER_CHANGE 100
#ifndef WATCHDOG_OTHER_SLEEP
#define WATCHDOG_OTHER_SLEEP 0
#endif
#ifdef H_BRIDGE_MOTOR
#define WATCHDOG_TOLERANCE 5
#else
#define WATCHDOG_TOLERANCE 20
#endif

I think I have to change something here to do my watchdog works.

#ifdef USENSOR <- Maybe this?

sxaris87, 2021-01-20:
without watchdog moving up or down with no error 1
also cant set favorites position neither min max

closed 2021-01-20 by sxaris87

sxaris87, 2024-02-11:
hello my friend. do you remember me?
i need your help again with skarsta project because i make an upgrade and something its wrong

yurazazik, 2024-02-11:
> hello my friend. do you remember me? i need your help again with skarsta project because i make an upgrade and something its wrong
> […](#)

Hello. Yes I remember you. What problem do you have in this time?

sxaris87, 2024-02-11:
oh my god, are you still there.

Listen,
at the beginning i has Rotary Encoder for 1-2 months and then after it damaged i replaced it with sonar (Ultrasonic Module HC-SR04 Distance Measuring Transducer Sensor).
With Sonar i was not happy because everything under the desk must be clear (even my feet was problem for sonar).
After that i saw this https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/sensor-hall.md
and i bought this https://www.aliexpress.com/item/1005005993720499.html?spm=a2g0o.productlist.main.1.24865b778xd16O&algo_pvid=b5568fc8-e025-4153-9f61-4c39d5a45994&algo_exp_id=b5568fc8-e025-4153-9f61-4c39d5a45994-0&pdp_npi=4%40dis%21EUR%2110.51%216.04%21%21%2179.76%2145.88%21%402101fb0d17076522674841555e3e79%2112000035234280891%21sea%21GR%211945225808%21&curPageLogUid=7bMjTudSR0gf&utparam-url=scene%3Asearch%7Cquery_from%3A
and i print the case.
Now when i connect Motor Hall Magnetic Encoder got Error 3 (wiring)

What i m doing wrong?
i made the connection with and without motor power terminals,
i reverse A - B and the results the same.
[cid:2609a950-71ee-4828-875e-1f4195b19add]

need to change the code from arduino? i dont remember doing anything from Rotary to Sonar.

Am i bought wrong Magnetic Encoder?

yurazazik, 2024-02-11:
Sorry but I didn't assemble this version with Sensor(

sxaris87, 2024-02-11:
Any ideas?
Or anyone who can help me?(an email)
Στάλθηκε από το iPhone μου

[1 more reply]