aenniw/ARDUINO #126: Cant set 1,2,3 favorite posisions or min/max if possible

sxaris87, 2021-01-20:
See the video please
https://1drv.ms/u/s!AnyS0ya9TatLnUxrhZwgvosx7SCf?e=GAGVYW

sxaris87, 2021-01-20:
ok i change
#define SENSOR_PIN0 2
#define SENSOR_PIN1 3
to 
#define SENSOR_PIN0 3
#define SENSOR_PIN1 2
and now it works

yurazazik, 2021-01-20:
> ok i change
> #define SENSOR_PIN0 2
> #define SENSOR_PIN1 3
> to
> #define SENSOR_PIN0 3
> #define SENSOR_PIN1 2
> and now it works

Can you write on my E-mail?

sxaris87, 2021-01-20:
sure. 
i think my encoder is not working good. not count well and now again not counting

yurazazik, 2021-01-20:
> sure.
> i think my encoder is not working good. not count well and now again not counting

Yes, your encoder is not working

sxaris87, 2021-01-20:
i don't want to print again because i don't have my own printer.
i will order an other encoder
pfff i have spend to many hour at this project till now.

yurazazik, 2021-01-20:
> i don't want to print again because i don't have my own printer.
> i will order an other encoder
> pfff i have spend to many hour at this project till now.

You don't need to print. You can use double-sided tape to secure the sensor.

sxaris87, 2021-01-20:
and the perfect spot to put it?
dont need an other code for this?

yurazazik, 2021-01-20:
> and the perfect spot to put it?
> dont need an other code for this?

I will help you with this

yurazazik, 2021-01-20:
Can you show you your power supply?

sxaris87, 2021-01-20:
my father had an other 12V at 5A and work fine

sxaris87, 2021-01-20:
and how this sensor work really? nothing rotating

yurazazik, 2021-01-20:
> and how this sensor work really? nothing rotating
Yeesss!

yurazazik, 2021-01-20:
> my father had an other 12V at 5A and works fine

Can you write a model this power suply?

yurazazik, 2021-01-20:
> > my father had an other 12V at 5A and works fine
> 
> Can you write a model this power supply?

yurazazik, 2021-01-20:
I think your encoder is not working because you did the wrong assembly.

sxaris87, 2021-01-20:
https://1drv.ms/u/s!AnyS0ya9TatLnU1dQ7zv1MlhoRRk

sxaris87, 2021-01-20:
i dont know what happened, the assembly was like diagramm. 
encoder     CLK to d2
and encoder DT to d3
after not working i change from platformio 
#define SENSOR_PIN0 2
#define SENSOR_PIN1 3
to
#define SENSOR_PIN0 3
#define SENSOR_PIN1 2

and it works but some times loses counting

yurazazik, 2021-01-20:
> https://1drv.ms/u/s!AnyS0ya9TatLnU1dQ7zv1MlhoRRk
thank you

sxaris87, 2021-01-20:
the encoder have 5 pins. 
CLK / DT/ SW / + / GND 
and i put it CLK -> D2 
                   DT  -> D3
                   SW -> empy
                    +  -> 5V
                 GND -> GND
m i right?

yurazazik, 2021-01-20:
> the encoder have 5 pins.
> CLK / DT/ SW / + / GND
> and i put it CLK -> D2
> DT -> D3
> SW -> empy
> + -> 5V
> GND -> GND
> m i right?

yes. It may be 2 problem encoder or your version programm

sxaris87, 2021-01-20:
2 problem encoder what you mean?
platformio version?

yurazazik, 2021-01-20:
> 2 problem encoder what you mean?
> platformio version?

Defective Encoder. Yes platformio

sxaris87, 2021-01-20:
> 
> 
> > 2 problem encoder what you mean?
> > platformio version?
> 
> Defective Encoder. Yes platformio

yes maybe defective, my platformIO version is this:
Version: 1.52.1 (system setup)
Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
Date: 2020-12-16T16:34:46.910Z
Electron: 9.3.5
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19042

sxaris87, 2021-01-20:
i order now  Distance Measuring Transducer Sensor like yours. 
how i replace the for the new sensor?

yurazazik, 2021-01-20:
> i order now Distance Measuring Transducer Sensor like yours.
> how i replace the for the new sensor?

[image]

sxaris87, 2021-01-20:
> 
> 
> > i order now Distance Measuring Transducer Sensor like yours.
> > how i replace the for the new sensor?
> 
> [image]

ah ok good semilar diagramm. and the code?

sxaris87, 2021-01-20:
> 
> 
> > sure.
> > i think my encoder is not working good. not count well and now again not counting
> 
> Yes, your encoder is not working. So I am using USensor! https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/sensor-sonar.md

is already inside the code of UltraSonicDistanceSensor?

yurazazik, 2021-01-20:
> > > sure.
> > > i think my encoder is not working good. not count well and now again not counting
> > 
> > 
> > Yes, your encoder is not working. So I am using USensor! https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/sensor-sonar.md
> 
> is already inside the code of UltraSonicDistanceSensor?

yes

sxaris87, 2021-01-20:
> > > > sure.
> > > > i think my encoder is not working good. not count well and now again not counting
> > > 
> > > 
> > > Yes, your encoder is not working. So I am using USensor! https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/sensor-sonar.md
> > 
> > 
> > is already inside the code of UltraSonicDistanceSensor?
> 
> yes

So, when the new SonicSensor arrives, have to replace the defective and that’s all?

yurazazik, 2021-01-20:
> > > > > конечно. 
> > > > > Я думаю, что мой кодировщик не работает. не считать хорошо и теперь снова не считая
> > > > 
> > > > 
> > > > Да, ваш кодировщик не работает. Так что пользуюсь USensor! https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/sensor-sonar.md
> > > 
> > > 
> > > уже находится внутри кода UltraSonicDistanceSensor?
> > 
> > 
> > да
> 
> Итак, когда прибудет новый датчик SonicSensor, придется заменить неисправный и все?

3

yurazazik, 2021-01-20:
> So, when the new SonicSensor arrives, have to replace the defective and that’s all?

Ofcourse

yurazazik, 2021-01-20:
But may be your encoder work too, may be this program version mistake

sxaris87, 2021-01-20:
> But may be your encoder work too, may be this program version mistake. Is this encoder work before?

I don’t know if this encoder work before.
Where can I look for a mistake?

yurazazik, 2021-01-20:
For this you need download platformio.ini one more time from site and replace it in your work folder

sxaris87, 2021-01-20:
ok i replace it

yurazazik, 2021-01-20:
> ok i replace it

tell me when you will finish

sxaris87, 2021-01-20:
ready. i paste the platformio.ini that i just download from the site to my peoject folder

yurazazik, 2021-01-20:
now open the project in VSC

sxaris87, 2021-01-20:
ok.

yurazazik, 2021-01-20:
[image]

yurazazik, 2021-01-20:
After that click here

[image]

sxaris87, 2021-01-20:
i follow you

yurazazik, 2021-01-20:
And choose here 
[image]

yurazazik, 2021-01-20:
After connect your arduino and compile project
[image]

yurazazik, 2021-01-20:
And will do this 
[image]

sxaris87, 2021-01-20:
https://ibb.co/tZ6FVdj

yurazazik, 2021-01-20:
> https://ibb.co/tZ6FVdj

What that? I cannot open this.

sxaris87, 2021-01-20:
> 
> 
> > https://ibb.co/tZ6FVdj
> 
> What that? I cannot open this.

you can open that? https://1drv.ms/u/s!AnyS0ya9TatLnU4tNlZV8F6ZEQx1?e=JvEW6C

sxaris87, 2021-01-20:
https://1drv.ms/u/s!AnyS0ya9TatLnU_OpM2X08jGpEwV?e=o6ErJ5

yurazazik, 2021-01-20:
I can now. So you compiled and you transfer program in arduino

yurazazik, 2021-01-20:
Connect power supply and try to work with table

sxaris87, 2021-01-20:
need to disconnect usb from pc to connect the power supply or it's ok to have them both connected?

sxaris87, 2021-01-20:
ok i uploaded and when i press up. Err1 on display

yurazazik, 2021-01-20:
> ok i uploaded and when i press up. Err1 on display

show me your configuration.h

sxaris87, 2021-01-20:
https://1drv.ms/u/s!AnyS0ya9TatLnVD15K0WcH2TlcnP?e=2nZsJr

yurazazik, 2021-01-20:
> https://1drv.ms/u/s!AnyS0ya9TatLnVD15K0WcH2TlcnP?e=2nZsJr

and MotorBridge.h

sxaris87, 2021-01-20:
https://1drv.ms/u/s!AnyS0ya9TatLnVHs8lMMYOrig9te?e=HCQn1m

yurazazik, 2021-01-20:
In MotorBridge.h change this like that
#define MIN_SPEED 255
#define MAX_SPEED 255
#define SPEED_STEP_DURATION 0

sxaris87, 2021-01-20:
moved almost 1cm and Er 1

yurazazik, 2021-01-20:
> In MotorBridge.h change this like that
> #define MIN_SPEED 255
> #define MAX_SPEED 255
> #define SPEED_STEP_DURATION 0
> 
> After this save file and compile and transfer in arduino

Did you try this?

[83 more replies]

closed 2021-01-23 by sxaris87