aenniw/ARDUINO #127: Usensor

yurazazik, 2021-01-22:
After how many cm should the information on the table height be displayed on the screen in the Urelay version and can this be changed or this parameter should be displayed in configuration.h? I output once every 2cm.

sxaris87, 2021-01-27:
@aenniw i need this too.

sxaris87, 2021-02-02:
Anything new? About the accuracy of sonar?

sxaris87, 2021-02-02:
in my case when it’s moving to a higher position it counts per 2cm but when it’s moving to a lower position it counts per 4cm. Who we can do it more accurate? @aenniw

yurazazik, 2021-02-02:
No 

> in my case when it’s moving to a higher position it counts per 2cm but when it’s moving to a lower position it counts per 4cm and this is happens because it’s moving faster when going down. Who we can do it more accurate? @aenniw

No, the fact is that your sonar is in the wrong place, you need to place it in the center of the table right behind the horizontal bar

yurazazik, 2021-02-02:
[image]

yurazazik, 2021-02-03:
> in my case when it’s moving to a higher position it counts per 2cm but when it’s moving to a lower position it counts per 4cm and this is happens because it’s moving faster when going down. Who we can do it more accurate? @aenniw

So?

sxaris87, 2021-02-03:
I m a little busy this days and I can’t make the transfer now

sxaris87, 2021-02-03:
So, nothing new about the accuracy of sonar from creator.

yurazazik, 2021-02-03:
I think he put this in to do list

yurazazik, 2021-02-04:
> So, nothing new about the accuracy of sonar from creator.

Did you transfer sonar?

sxaris87, 2021-02-04:
No, maybe at the next days

sxaris87, 2021-02-06:
Where can I put it to work fine? 
At the center if I m seating the sonar at 80% hits to my legs 1st 
[image]

yurazazik, 2021-02-08:
> Where can I put it to work fine?
> At the center if I m seating the sonar at 80% hits to my legs 1st
> [image]

You have to remove all cables and metallic parts and put sensor on this place.

sxaris87, 2021-02-08:
This is not possible. Need an other way

yurazazik, 2021-02-08:
> This is not possible. Need an other way

Another way is just to move the sensor and check and so on until you find the correct position and remember that the sensor sends sound not only downward but in a certain radius, therefore, there should be nothing about 10-20 cm in radius on the floor and around the sensor.

aenniw, 2021-02-09:
@yurazazik currently min change to display is 1cm  that is checked each 125ms, configuration extracted and added description see #131

aenniw, 2021-02-09:
@sxaris87 the possition of usensor is quite tricky as it depends if you have table near the walls or even on the floor uneven/carpet... thus you need to play with it quite a lot

[image]

yurazazik, 2021-02-09:
> @yurazazik currently min change to display is 1cm that is checked each 125ms, configuration extracted and added description see #131

Thank you so much! And what is it file do this check?

aenniw, 2021-02-10:
Check is performed in [Motor.cpp](https://github.com/aenniw/ARDUINO/blob/master/skarsta/lib/motor/Motor.cpp#L166) and its config is in [Motor.h](https://github.com/aenniw/ARDUINO/blob/master/skarsta/lib/motor/Motor.h#L32)

yurazazik, 2021-02-11:
> Check is performed in [Motor.cpp](https://github.com/aenniw/ARDUINO/blob/master/skarsta/lib/motor/Motor.cpp#L166) and its config is in [Motor.h](https://github.com/aenniw/ARDUINO/blob/master/skarsta/lib/motor/Motor.h#L32)

Thank you for your reply. I rebuilt the sensor by 10 mm, then by 5 mm, but it began to work much worse than the original version, I returned everything to 20mm by the end

sxaris87, 2021-02-11:
> Thank you for your reply. I rebuilt the sensor by 10 mm, then by 5 mm, but it began to work much worse than the original version, I returned everything to 20mm by the end

What are this numbers? Which line? What do you mean you rebuilt the sensor by 10mm?

yurazazik, 2021-02-11:
> What are this numbers? Which line? What do you mean you rebuilt the sensor by 10mm?

If you carefully read the posts above, Aenniw kindly told where in the program to change the sensor settings and how to make it show information on the screen every 10 mm for the standard 20mm space

sxaris87, 2021-02-11:
SENSOR_DELTA_ON     10mm
SENSOR_DELTA_OFF    75mm?
This things you mean you  change?

yurazazik, 2021-02-11:
> SENSOR_DELTA_ON 10mm
> SENSOR_DELTA_OFF 75mm?
> This things you mean you change?

yes

sxaris87, 2021-02-11:
> > SENSOR_DELTA_ON 10mm
> > SENSOR_DELTA_OFF 75mm?
> > This things you mean you change?
> 
> yes

And now what is your settings for sensor delta on/off ? Default it’s 10/75 by Aenniw

yurazazik, 2021-02-11:
> > > SENSOR_DELTA_ON 10mm
> > > SENSOR_DELTA_OFF 75mm?
> > > This things you mean you change?
> > 
> > 
> > yes
> 
> And now what is your settings for sensor delta on/off ? Default it’s 10/75 by Aenniw

When i use - Default it’s 10/75 nothing change. It work as before every 2cm

sxaris87, 2021-02-11:
And you find the way to make it count every 1cm?

yurazazik, 2021-02-11:
> And you find the way to make it count every 1cm?

It works very bad. Numbers are constantly jumping here and there. For example, it was just 103 cm then 102 and then again 103 and again 102

closed 2021-03-20 by github-actions[bot]