aenniw/ARDUINO #129: Are sensors required? rot vs hall vs sonar benefits?

trancefam, 2021-01-30:
Hello,

Thank you very much for the project. I am about to order all of the necessary parts for the relay setup. It is not clear to me if all 3 of the below sensors are required? If only one is required, are there advantages in choosing one or the other?

- sensor-rot - relative position
- sensor-hall - relative position
- sensor-sonar - absolute position

yurazazik, 2021-02-03:
Only one is required and the simplest option in my opinion is sensor-sonar

aenniw, 2021-02-09:
Yes only one is required as stated above, while by default `sensor-rot` is embedded in [cases v1](https://github.com/aenniw/ARDUINO/tree/master/skarsta#electronic-cases). 

Sensors mentioned in [sensors-cases](https://github.com/aenniw/ARDUINO/tree/master/skarsta#sensors-cases) chapter are prestep for #130 however you can use them also with v1, just need to place them out....

`sensor-hall` and `sensor-sonar` are good options, however for `sensor-sonar` you need to have space under table to avoid accidental trigerring of it...

closed 2021-03-17 by github-actions[bot]