aenniw/ARDUINO #134: Model sizes to small?

Counfhou, 2021-03-03:
Hi, 
First of all, amazing project and great documentation.
I started printing the models to build this but I noticed with mutliple parts now that they are all a tiny bit to small, going from the sonar encasing, to the motor-pulley, rotary pulley etc. Somehow on all of them the openings are like a mm or so to small so nothing fits. I first thought it would be one off so I printed a couple to test, am I missing something in my cura settings you have enabled or?

yurazazik, 2021-03-17:
> Hi,
> First of all, amazing project and great documentation.
> I started printing the models to build this but I noticed with mutliple parts now that they are all a tiny bit to small, going from the [...]

And when you print a proof cube, does it have 2 by 2 and 2 cm edges?

Counfhou, 2021-03-17:
I started diging into this but it seems I have like an adhesion problem which I am hoping to solve with a better bed. The holes on the bottom, directly to the buildplate are always slightly off/too small. Never having printed either on the bottom/so tight I just did not notice this before. But clearly it is my printer.

yurazazik, 2021-03-18:
> I started diging into this but it seems I have like an adhesion problem which I am hoping to solve with a better bed. The holes on the bottom, directly to the buildplate are always slightly off/too [...]

Try to use for print PetG plastic with 235 degrees celsius and to use glue pencil and turn off hot bed. https://www.youtube.com/watch?v=xS3MWUkl6FE

closed 2021-04-24 by github-actions[bot]