aenniw/ARDUINO #144: Female 8 pin socket as gnd/5v slot

Counfhou, 2021-07-23:
Hi,
Silly question probably but how did you do the 8 pin sockets as sltos for the gnd/5v?
I thought you would just crimp and ad cables at the bottom and solder everyting together, but it seems as if there are no wire in your picture. Did you use a different method taking less place to connect them al somehow?

thx

aenniw, 2021-08-10:
Which picture do you mean? Could you link the docs steps that you mean?

Counfhou, 2021-08-11:
[image]

Thx!

aenniw, 2021-08-13:
I see, I've just bridged pins on `8-pin socket` inserted it into mount and for the input connected jumper wire from `buck-converter`

I'll extends the docs with more robust explanation regarding it....

PR #130 (2021-08-15): Add modular skarsta designs
- deprecated aio skarsta designs
- added separate case for motor models
- added docs for motor case assembly
- added separate case for electronics (relays/h-bridge)
- added docs for electronics assembly
- fixed typos, refactored docs

closes #144
closed 2021-08-15 by aenniw