aenniw/ARDUINO #148: Wrong coupler size and link

zoltan-kiss-hungary, 2021-11-05:
The motor case page says you need the 6mm-10mm coupler:

https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/case_motor-v2.md

But in reality you need the 6/7 version. And on the ebay link they no longer sell that size. I'm still thinking how to downsize the 8mm hole on mine, because a new one would take another 2 month to arrive again :(

mikekay1, 2021-12-02:
Do you know anyone with a 3D Printer?

zoltan-kiss-hungary, 2021-12-04:
I do, but the 3D printed coupler wore out while I was setting up the table, so I ordered one, I'll post the link when it arrives and works out.

mikekay1, 2021-12-15:
What motor did you source man, just wanna make sure I get the right one...

zoltan-kiss-hungary, 2021-12-15:
I have this one:

https://www.aliexpress.com/item/32881467792.html

But it's not available now. It supposed to be the same kind as the one linked in the introduction text of this project. At least based on the photos and description.

closed 2022-01-20 by github-actions[bot]