aenniw/ARDUINO #149: Table not moving (H-Bridge version)

Josch182, 2021-11-13:
Hello, first of all thank you very much for the detailed instruction. Had some issues but the main issues was the thing between my ears.

Anyhow, a burned and screaming arduino nano (overvoltage by mistake) later, I finally got that thing working - almost atleast.

Without the table connected to the motor, everything works fine until the error 2 appears because of the missing load.
After connecting the the table with a hexshaft to the coupling/motor and trying to go up and down, the motor tries hard to move the table but error 1 appears after 1-2 sec. It seems the motor is not strong enough to drive the table.
All parts used are the ones suggested in the instruction.

Parts of interested used:

- Noname from local electronics store 12v 5A power supply

- Geared Motor XD-37GB555 15 W 12V 500RPM (not 180RPM, is that a problem?)
https://www.amazon.de/gp/aw/d/B07RSWZYDL?psc=1&ref=ppx_pop_mob_b_asin_title

Any clue whats going wrong? Help would be very appreciated as i really wish to get this project over the finish line.

Thanks in advance!

mikekay1, 2021-12-02:
I am about to embark on this project and just about to click go on my shopping cart BOM (Bill of Materials)...

I think the 500RPM doesn't provide the torque needed to move the table. The higher the speed the less the torque... since it is geared for speed and not torque you will need to buy again. 

The interesting part is the banggood listed product says the torque is 90kg*cm which is massive! I am pretty sure that it is wrong because there isn't a single motor on ebay or aliexpress with those specs. I too am trying to source the right motor... I am building the relay version, which has the click on and off -- maybe H Bridge is better still contemplating. 

Let me know how you made out!

Josch182, 2021-12-04:
Hey Mike!
Thank you very much for your answer. I already almost gave up on hope for someone to reply to be honest...
In the meantime I ordered a different motor with less rpm and therfore more torque (Geared 12v Motor XC37GB30) but had no luck this this one either. Tabletop is still not moving in any direction, neither up nor down. Always Error 1. I really don't what the problem is. Moving it by hand with the crank is pretty easy, not very easy but I think it works totally normal.

The 90kg torque from the banggood motor sound really to good to be true. I did't order that one as it would have taken to long to ship it to my location, so went with amazon. But yeah, the table is still not moving.

Did you decide on which route to go h-brige or relay in the meantime?

mikekay1, 2021-12-04:
Im still not sure, I am more familiar with relays then HBridges... But the click might start to get to me :)

30RPM would have had the rquired torque one would think! Which rotational sensor did you use? And did you try just running straight power into the motor does it move the table? Because current matters are you sure you got the 5a power supply? I have a spare motor and PSU at home that I will see if works as a test and ppst back for you. If you want we can pop on discord or something and figure it out again I need to make sure I order everything correctly so using your material will help me iron out exactly what I need. Ill post back with pics soon!

Josch182, 2021-12-30:
hey mike,
I'm close to the edge of giving up that project...

Your suggestion of running straight power into the motor was pretty helpful. The 30rpm moved the table nicely that way but was obviously pretty slow, too slow for my taste so I ordered a 200rpm motor that I found the other day. Like the 30rpm one it worked pretty good with straight power as well so I assembled everything back together with the new 200rpm motor. After a few cm of table movement error 1 appeared so I adjusted the watchdog sleep settings to 5000 but error 1 was still coming up. I then disabled the watchdog completely as the whole thing is fused anyway. After that the table was finally moving but the motor was working very hard and also seemed to lose power as the movment slowed down signifcantly  the higher the table got. So I thought it was the power supply to blame as it was a no name 12v 5A one. So I ordered a new one with 12v 10A, but guess what - made no difference - the slowing down problem still persists. Arrr...

Also the memory function is not working properly. After saving bottom and top position with the tilde button, the up an down buttons are not working anymore. Replacing the KY-040 rotary encoder also didn't changed anything.

I have no idea now how to fix these problems anymore. May you have? please...

Things i already tried with no success so far: (found in different postings here)
- leveling the table
- H-Bridge speed Min/Max set to 255
- Replacing KY-040 rotary encoder
- Stronger power supply with 10A

One thing I didn't try is to grease the tables mechanics as it works pretty easy with the crank. Do you know how to open tha plastic covers of the mechanics?

I read somewhere that the H-Bridge version is not as strong as the relay version is that true?

I'm curious how you project is going...

mikekay1, 2021-12-30:
Hey man,

I havent gotten to this part of the project just yet and havent ordered 
anything.

Im glad some of my suggestions were helpful! If running straight power to 
the motor works and works well then some else is not working or adding 
"drag" to the motor. Slow is slow but you have to remember that RPM and 
torque are opposites of each other unless you upgrade the power going into 
the motor ie 24v, or upgrade the motor itself to a bigger one like a 
stepper or something with some massive crank.

One thing I forgot to ask is does this all happen fully assembled?

I know you want it controlled and have a little led control panel but a 
drill, those home use 12v cheapy ones from ebay or amazon work very well! 
Cut the handle off the original IKEA crank, and then insert it into the 
chuck. You can also use a rod with hex welded onto the end. Key here is 
make sure everything is perfect straight, and I would change out the 
plastic guide at the front of the table yo something with a skateboard 
bearing and then lightly pull the trigger.

Ill probably have more info as I get more into this project but also might 
fork aomething of my own. With the developer not even chiming in -- we are 
all just guessing here...

Hope that helps a bit.

sv01a, 2022-01-15:
I had same issue with KY-040. After swapping CLK and DT pins all works fine. You could try to swap it on encoder or simply in config SENSOR_PIN0 and SENSOR_PIN1

closed 2022-02-20 by github-actions[bot]