aenniw/ARDUINO #150: Rotary sensor STL missing

keesvanbemmel, 2022-01-16:
Hi! 

The rotary mount STL seems missing.

On this page: https://github.com/aenniw/ARDUINO/blob/master/skarsta/docs/sensor-rot.md

Rot-mount STL links to hall-mount.

Should I use that one?

trewbene, 2022-02-26:
Unfortunately I can confirm that the linked hall-mount does not seem to match the required rot-mount.
It would be great if the correct STL could be uploaded and linked in the documentation.

@aenniw 
Regardless, I would like to say that I am very excited about the work that went into this project. Thank you very much!

aenniw, 2022-03-01:
@keesvanbemmel The link in the `docs` is correct despite that it points to `hal-mount.stl` as the  `mount` parts are identical for both of them, and as long as `part id`matches they should be compatible.... 

@trewbene what parts doesn't match each other? If its regarding the `mount` <-> `cover` it should be caused by tolerances that could be tweaked...

trewbene, 2022-03-15:
Sorry for the late reply. Somehow it didn't seem to fit from the STL files.
Now that I have printed the parts, they fit together as you described.

closed 2022-04-03 by aenniw